pibooth/pibooth #96: Hide the last taken picture

sblanc27, 2019-11-05:
On the main screen, after a first picture has been taken, the last picture is displayed for another print.
It would be really great to have to functionality to hide this last picture taken or to have a timer.

Note: I could do this by commenting the following lines on the window.py file.

```
    def show_intro(self, pil_image=None, with_print=True):
        """Show introduction view.
        """
        self._capture_number = (0, self._capture_number[1])
        if with_print and pil_image:
            self._update_background(background.IntroWithPrintBackground(self.arrow_location, self.arrow_offset))
        else:
            self._update_background(background.IntroBackground(self.arrow_location, self.arrow_offset))

 #       if pil_image:
 #           self._update_foreground(pil_image, self.RIGHT)

```

werdeil, 2019-11-05:
I'm not really fan of hiding the last picture, can you give me a use case where it is problematic?

sblanc27, 2019-11-05:
Let say for a self-service booth on public event on which the person does not know the previous one.

anxuae, 2019-11-05:
Hi,

An option could be added to hide the picture on first screen (for instance `[WINDOW][show_last]`).
In case of True, the `IntroBackground` class should always be used. Maybe the text can be centred on screen?

anxuae

werdeil, 2019-11-05:
I've pushed a first implementation with a timeout in commit 0a7a3e7d16af3c3e909a1ccc0fdacd78a9acc141 (and I have named the option [WINDOW][final_image_delay])

If the delay is 0, the image is never shown, if the delay is -1, the image is always shown

I'm not convinced to center the text on the screen in case there is no image though, because in this case when the timeout is reached the text will move.

werdeil, 2019-11-06:
If think everything is correctly handled in 35f620e.

I'm closing the issue it will be available in next release

closed 2019-11-06 by werdeil