pibooth/pibooth #97: Setting menu translation

debouttek, 2019-11-10:
Hi
Is it possible to translate the SETTINGS MENU? pibooth --translate is only for the users interface.
Do I need to look into python scripts?

Thanks

anxuae, 2019-11-15:
This is hard to manage the menu size depending on the language.
@werdeil what do you think about this enhancement?

werdeil, 2019-11-15:
Menu is only a part of the config options, the ones that we thought interesting to have directly in a GUI.

For me, I see the config keys behind and it makes no sense to translate config keys.
Moreover I don't think it is a good idea to translate only part of the config, and translating the whole config is too much.

So i won't do it.

werdeil, 2020-02-02:
As explained in the previous comment, I don't think we should translate the settings menu. I'm thus closing the issue.

closed 2020-02-02 by werdeil