pibooth/pibooth #98: Installation Failed

DEhrhardt, 2019-11-14:
Hi,

I followed all steps of the installation instructions, I just skipped installing cups.

Then I installed pibooth with the following command and got the following error:

[](pi@raspberrypi:~ $ sudo pip3 install pibooth[dslr]
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting pibooth[dslr]
  Downloading https://files.pythonhosted.org/packages/39/17/2069e27baefd8998487d16350e5c3bf5b24fc8f33e5c0bcfea66915fa5a8/pibooth-1.2.0-py2.py3-none-any.whl (605kB)
    100% |████████████████████████████████| 614kB 424kB/s 
Requirement already satisfied: psutil>=5.5.1 in /usr/lib/python3/dist-packages (from pibooth[dslr]) (5.5.1)
Collecting Pillow>=6.1.0 (from pibooth[dslr])
  Downloading https://files.pythonhosted.org/packages/5b/bb/cdc8086db1f15d0664dd22a62c69613cdc00f1dd430b5b19df1bea83f2a3/Pillow-6.2.1.tar.gz (37.7MB)
    100% |████████████████████████████████| 37.7MB 7.3kB/s 
Requirement already satisfied: RPi.GPIO>=0.6.3; platform_machine >= "armv0l" and platform_machine <= "armv9l" in /usr/lib/python3/dist-packages (from pibooth[dslr]) (0.7.0)
Requirement already satisfied: Pygame>=1.9.4 in /usr/lib/python3/dist-packages (from pibooth[dslr]) (1.9.4.post1)
Collecting pygame-menu>=2.0.4 (from pibooth[dslr])
  Downloading https://files.pythonhosted.org/packages/0d/67/24d109a07c802ff41e47a5bac392708853bbdfe33df428289358ee1f0e8a/pygame_menu-2.0.4-py3-none-any.whl (544kB)
    100% |████████████████████████████████| 552kB 475kB/s 
Requirement already satisfied: picamera>=1.13; platform_machine >= "armv0l" and platform_machine <= "armv9l" in /usr/lib/python3/dist-packages (from pibooth[dslr]) (1.13)
Collecting gphoto2>=2.0.0; extra == "dslr" (from pibooth[dslr])
  Downloading https://files.pythonhosted.org/packages/6a/2e/8d7b712bc3de6552dd599b57ac400d817c64b4e5630de907981efba460bd/gphoto2-2.0.0.tar.gz (7.2MB)
    100% |████████████████████████████████| 7.2MB 38kB/s 
Collecting pyperclip (from pygame-menu>=2.0.4->pibooth[dslr])
  Downloading https://www.piwheels.org/simple/pyperclip/pyperclip-1.7.0-py3-none-any.whl
Building wheels for collected packages: Pillow, gphoto2
  Running setup.py bdist_wheel for Pillow ... error
  Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-3cmt4ok7/Pillow/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-j0q4s0ii --python-tag cp37:
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-armv7l-3.7
  creating build/lib.linux-armv7l-3.7/PIL
  copying src/PIL/PcdImagePlugin.py -> build/lib.linux-armv7l-3.7/PIL
  copying src/PIL/XpmImagePlugin.py -> build/lib.linux-armv7l-3.7/PIL
  copying src/PIL/ContainerIO.py -> build/lib.linux-armv7l-3.7/PIL
  copying src/PIL/PpmImagePlugin.py -> build/lib.linux-armv7l-3.7/PIL
  copying src/PIL/ImageWin.py -> build/lib.linux-armv7l-3.7/PIL
  copying src/PIL/GimpGradientFile.py -> build/lib.linux-armv7l-3.7/PIL
  copying src/PIL/CurImagePlugin.py -> build/lib.linux-armv7l-3.7/PIL
  copying src/PIL/ExifTags.py -> build/lib.linux-armv7l-3.7/PIL
  copying src/PIL/ImageEnhance.py -> build/lib.linux-armv7l-3.7/PIL
  copying src/PIL/IptcImagePlugin.py -> build/lib.linux-armv7l-3.7/PIL
  copying src/PIL/ImageDraw.py -> build/lib.linux-armv7l-3.7/PIL
  copying src/PIL/PaletteFile.py -> build/lib.linux-armv7l-3.7/PIL
  copying src/PIL/Hdf5StubImagePlugin.py -> build/lib.linux-armv7l-3.7/PIL
  copying src/PIL/PngImagePlugin.py -> build/lib.linux-armv7l-3.7/PIL
  copying src/PIL/FpxImagePlugin.py -> build/lib.linux-armv7l-3.7/PIL
  copying src/PIL/JpegPresets.py -> build/lib.linux-armv7l-3.7/PIL
  copying src/PIL/ImagePath.py -> build/lib.linux-armv7l-3.7/PIL
  copying src/PIL/EpsImagePlugin.py -> build/lib.linux-armv7l-3.7/PIL
  copying src/PIL/ImageFilter.py -> build/lib.linux-armv7l-3.7/PIL
  copying src/PIL/ImageColor.py -> build/lib.linux-armv7l-3.7/PIL
  copying src/PIL/XbmImagePlugin.py -> build/lib.linux-armv7l-3.7/PIL
  copying src/PIL/PcxImagePlugin.py -> build/lib.linux-armv7l-3.7/PIL
  copying src/PIL/TgaImagePlugin.py -> build/lib.linux-armv7l-3.7/PIL
  copying src/PIL/GbrImagePlugin.py -> build/lib.linux-armv7l-3.7/PIL
  copying src/PIL/PdfParser.py -> build/lib.linux-armv7l-3.7/PIL
  copying src/PIL/__main__.py -> build/lib.linux-armv7l-3.7/PIL
  copying src/PIL/PyAccess.py -> build/lib.linux-armv7l-3.7/PIL
  copying src/PIL/ImageChops.py -> build/lib.linux-armv7l-3.7/PIL
  copying src/PIL/IcnsImagePlugin.py -> build/lib.linux-armv7l-3.7/PIL
  copying src/PIL/Jpeg2KImagePlugin.py -> build/lib.linux-armv7l-3.7/PIL
  copying src/PIL/MpegImagePlugin.py -> build/lib.linux-armv7l-3.7/PIL
  copying src/PIL/_version.py -> build/lib.linux-armv7l-3.7/PIL
  copying src/PIL/ImImagePlugin.py -> build/lib.linux-armv7l-3.7/PIL
  copying src/PIL/FontFile.py -> build/lib.linux-armv7l-3.7/PIL
  copying src/PIL/Image.py -> build/lib.linux-armv7l-3.7/PIL
[... 298 lines cut ...]
      File "/usr/lib/python3.7/distutils/command/build_ext.py", line 340, in run
        self.build_extensions()
      File "/tmp/pip-install-3cmt4ok7/Pillow/setup.py", line 687, in build_extensions
        raise RequiredDependencyException(f)
    __main__.RequiredDependencyException: jpeg
    
    During handling of the above exception, another exception occurred:
    
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-3cmt4ok7/Pillow/setup.py", line 903, in <module>
        raise RequiredDependencyException(msg)
    __main__.RequiredDependencyException:
    
    The headers or library files could not be found for jpeg,
    a required dependency when compiling Pillow from source.
    
    Please see the install instructions at:
       https://pillow.readthedocs.io/en/latest/installation.html
    
    
    
    ----------------------------------------
  Can't roll back Pillow; was not uninstalled
Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-3cmt4ok7/Pillow/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-rs43itv6/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-install-3cmt4ok7/Pillow/)

anxuae, 2019-11-15:
Hi @DEhrhardt ,

This is because, pip try to install the last version of Pillow (=6.2.1), but this version is not yet available as binary package. Thus it tries to install form source, but compilation fails due to missing dependencies on your raspberrypi.

To bypass it, you can first install Pillow v6.1.0:
``
sudo pip install Pillow==6.1.0
``

DEhrhardt, 2019-11-15:
Thanks for your reply @anxuae 

I tried this, but received the same error.

I figured out that I had to install libjpeg first. I've done it this way:

`sudo apt-get install libjpeg8-dev`

Then the pibooth installation worked.

closed 2019-11-15 by DEhrhardt