pibooth/pibooth #100: Neopixel Support

jo-ei, 2019-12-12:
**Is your feature request related to a problem? Please describe.**
Hi, my idea it would be a support for driving a neopixel ring as in #39.
This could be attached around the hole for the camera, to get the people to look there.

**Describe the solution you'd like**
Support to configure a neopixel ring with different statuses.
1.) Idel
2.) Countdown
3.) Photo
4.) Printing

My desired config for this statuses  
1.) Rainbow slow motion
2.) red LED Clock remove pixel clockwise from 100% to 0%
3.) 100% white as light
4.) Off OR 1 dot rotation

Every Option should be configurable 
- LED MODE
- LED COUNT
- LED BRIGHTNESS
- LED SPEED

**Describe alternatives you've considered**
Give these 4 statuses (+ time for countdown) to a ESP8266 for external neopixel control and reduce complexity for application.

Thanks

anxuae, 2019-12-13:
Hi jo-ei,

Integrating the Neopixel could be a great improvement.
HoweverI don't know how it's work and I don't have a Neopixel ring to develop and test.

Maybe we can introduce a plugin system (eg https://pypi.org/project/pluggy ) to let the possibility to integrate custom functions at specific steps of the pibooth process.

anxuae

werdeil, 2019-12-15:
I like the idea of a plugin or hook system.

anxuae, 2020-03-18:
Hi @jo-ei ,

Do you have an example of a python script which is driving a ne-pixel ring?
I would like to do some test with the ``pibooth``plugin system.

thanks
anxuae

Aurel-Charles, 2020-03-18:
here, i'm not sure but I think that these have different GPIO mode and library: 

https://github.com/jgarff/rpi_ws281x/tree/master/python/examples

https://github.com/adafruit/Adafruit_CircuitPython_NeoPixel/blob/master/examples

but again i'm not sure about those differences ! 
hope that can help! (ho .... and so far great general updates 👍 )

anxuae, 2020-03-24:
New plugin system will let you the possibility to add custom code to drive other GPIO.
Will be available in next `pibooth` major release (date to be planned).
See plugins example at https://github.com/werdeil/pibooth/blob/master/docs/plugin.rst .

closed 2020-03-24 by anxuae