pibooth/pibooth #102: endless image to buffer added

jo-ei, 2019-12-14:
**Describe the bug**
endless add image to buffer after a multi picture with webcam

**Steps to reproduce the bug**
 Take a multi shot

**Logs**
```
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 441 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 459 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 480 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 480 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tEXt' 499 25
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 536 8192
[ INFO    ] pibooth           : +< Creating the final picture
[ INFO    ] pibooth           : |  +< Use OpenCvPictureMaker to create background
[ DEBUG   ] pibooth           : |  +> took 0.335 seconds
[ INFO    ] pibooth           : |  +< Use OpenCvPictureMaker to concatenate images
[ DEBUG   ] pibooth           : |  +> took 1.117 seconds
[ INFO    ] pibooth           : |  +< Use OpenCvPictureMaker to assemble final image
[ DEBUG   ] pibooth           : |  +> took 0.095 seconds
[ INFO    ] pibooth           : |  +< Use OpenCvPictureMaker to draw texts
[ DEBUG   ] pibooth           : |  +> took 0.057 seconds
[ DEBUG   ] pibooth           : +> took 3.243 seconds
[ INFO    ] pibooth           : +< Save image '/home/pi/Pictures/pibooth/2019-12-14-21-09-58_pibooth.jpg'
[ DEBUG   ] pibooth           : +> took 0.435 seconds
[ INFO    ] pibooth           : +< Asyncronously generate pictures for animation
[ DEBUG   ] pibooth           : +> took 0.237 seconds
[ DEBUG   ] pibooth           : Activate state 'finish'
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iCCP' 41 388
[ DEBUG   ] PIL.PngImagePlugin: iCCP profile name b'ICC profile'
[ DEBUG   ] PIL.PngImagePlugin: Compression method 0
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 441 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 441 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 459 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 480 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 480 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tEXt' 499 25
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 536 8192
[ INFO    ] pibooth           : |  +< Use PilPictureMaker to create background
[ DEBUG   ] pibooth           : |  +> took 0.101 seconds
[ INFO    ] pibooth           : |  +< Use PilPictureMaker to concatenate images
[ INFO    ] pibooth           : |  +< Use PilPictureMaker to create background
[ DEBUG   ] pibooth           : |  +> took 0.111 seconds
[ INFO    ] pibooth           : |  +< Use PilPictureMaker to concatenate images
[ INFO    ] pibooth           : |  +< Use PilPictureMaker to create background
[ DEBUG   ] pibooth           : |  +> took 0.170 seconds
[ INFO    ] pibooth           : |  +< Use PilPictureMaker to concatenate images
[ INFO    ] pibooth           : |  +< Use PilPictureMaker to create background
[ DEBUG   ] pibooth           : |  +> took 0.112 seconds
[ INFO    ] pibooth           : |  +< Use PilPictureMaker to concatenate images
[ DEBUG   ] pibooth           : Activate state 'wait'
[ DEBUG   ] pibooth           : |  +> took 1.764 seconds
[ INFO    ] pibooth           : |  +< Use PilPictureMaker to assemble final image
[ DEBUG   ] pibooth           : |  +> took 0.000 seconds
[ INFO    ] pibooth           : |  +< Use PilPictureMaker to draw texts
[ DEBUG   ] pibooth           : |  +> took 0.066 seconds
[ DEBUG   ] pibooth           : |  +> took 1.750 seconds
[ INFO    ] pibooth           : |  +< Use PilPictureMaker to assemble final image
[ DEBUG   ] pibooth           : |  +> took 0.000 seconds
[ INFO    ] pibooth           : |  +< Use PilPictureMaker to draw texts
[ DEBUG   ] pibooth           : |  +> took 0.076 seconds
[ DEBUG   ] pibooth           : |  +> took 1.813 seconds
[ INFO    ] pibooth           : |  +< Use PilPictureMaker to assemble final image
[ DEBUG   ] pibooth           : |  +> took 0.000 seconds
[ INFO    ] pibooth           : |  +< Use PilPictureMaker to draw texts
[ DEBUG   ] pibooth           : |  +> took 0.093 seconds
[ DEBUG   ] pibooth           : |  +> took 1.763 seconds
[ INFO    ] pibooth           : |  +< Use PilPictureMaker to assemble final image
[ DEBUG   ] pibooth           : |  +> took 0.000 seconds
[ INFO    ] pibooth           : |  +< Use PilPictureMaker to draw texts
[ DEBUG   ] pibooth           : |  +> took 0.106 seconds
[ DEBUG   ] pibooth           : Add to buffer the image '1665072976'
[ DEBUG   ] pibooth           : Add to buffer the image '1664412048'
[ DEBUG   ] pibooth           : Add to buffer the image '1664299600'
[ DEBUG   ] pibooth           : Add to buffer the image '1664808880'
[ DEBUG   ] pibooth           : Add to buffer the image '1665072976'
[ DEBUG   ] pibooth           : Add to buffer the image '1664412048'
[ DEBUG   ] pibooth           : Add to buffer the image '1664299600'
[ DEBUG   ] pibooth           : Add to buffer the image '1664808880'
[ DEBUG   ] pibooth           : Add to buffer the image '1665072976'
[ DEBUG   ] pibooth           : Add to buffer the image '1664412048'
[ DEBUG   ] pibooth           : Add to buffer the image '1664299600'
[ DEBUG   ] pibooth           : Add to buffer the image '1664808880'
[ DEBUG   ] pibooth           : Add to buffer the image '1665072976'
[ DEBUG   ] pibooth           : Add to buffer the image '1664412048'
[ DEBUG   ] pibooth           : Add to buffer the image '1664299600'
[ DEBUG   ] pibooth           : Add to buffer the image '1664808880'
[ DEBUG   ] pibooth           : Add to buffer the image '1665072976'
[ DEBUG   ] pibooth           : Add to buffer the image '1664412048'
[ DEBUG   ] pibooth           : Add to buffer the image '1664299600'
[ DEBUG   ] pibooth           : Add to buffer the image '1664808880'
[ DEBUG   ] pibooth           : Add to buffer the image '1665072976'
[ DEBUG   ] pibooth           : Add to buffer the image '1664412048'
[ DEBUG   ] pibooth           : Add to buffer the image '1664299600'
[ DEBUG   ] pibooth           : Add to buffer the image '1664808880'
[ DEBUG   ] pibooth           : Add to buffer the image '1665072976'
[ DEBUG   ] pibooth           : Add to buffer the image '1664412048'
[ DEBUG   ] pibooth           : Add to buffer the image '1664299600'
[ DEBUG   ] pibooth           : Add to buffer the image '1664808880'
[ DEBUG   ] pibooth           : Add to buffer the image '1665072976'
[ DEBUG   ] pibooth           : Add to buffer the image '1664412048'
[ DEBUG   ] pibooth           : Add to buffer the image '1664299600'
```

**Environment**
 - OS: Raspbian Buster
 - Camera: Microsoft Corp. LifeCam VX-2000
 - Pibooth version: 1.2.0

anxuae, 2019-12-16:
Hi @jo-ei ,

Thanks for you feedback. This is in fact not a bug, I think that you have enabled the `animated pictures` feature during the wait state?
 The `buffer` of the window is used to keep reference to the currently displayed picture. Because of animated picture, the buffered image is endless changed.

But I can manage to make log massages less verbose or more explicite.

anxuae

werdeil, 2020-02-02:
As @anxuae said it is the intended logging level when you use the animated picture feature.

I'm closing the issue as it is not an bug.

closed 2020-02-02 by werdeil