pibooth/pibooth #104: show result photo fullscreen

jo-ei, 2019-12-15:
**Is your feature request related to a problem? Please describe.**
if you don't have a printer connected, it would be nice if you could display the finished picture in full screen

**Describe the solution you'd like**
config option to display only the picture

werdeil, 2019-12-15:
Hi,

Thanks for the feedback from your other issues I guess that your are generating a landscape result photo. This is also a limitation of landscape photos, portrait ones are taking the whole height of the screen.

Your solution is interesting but doesn't address the case of users with a printer and a photo in landscape mode.

jo-ei, 2019-12-15:
Hi @werdeil ,
thanks for the info.

But I think that would be very useful, because most of the photo boxes take pictures in landscape format for a group photo. If you don't have a direct print, people would like to see the photo at least directly at the time of shooting.

anxuae, 2020-03-21:
@werdeil in the future version, the print screen will be used also to "forget" the picture.
Thus this state could become mandatory, and we can hide the `print arrow` if no printer available.

mmk8283, 2020-10-24:
I would also like a way of displaying the finished collage full screen without the "take a photo" and arrow on the side of the screen. is there any progress on this issue? 

Thank you!

werdeil, 2020-10-26:
Hi @mmk8283 can you confirm that you are not using a printer? because the final photo is already displayed at two step: print and wait.

 One change we can think of is to display to the print screen even if there is no printer (without print button).

mmk8283, 2020-10-26:
correct, I am not using a printer.

I would like to thank you by the way, this is a very nice photobooth. Thank
you for all your hard work!

mmk8283, 2020-10-27:
> Hi @mmk8283 can you confirm that you are not using a printer? because the final photo is already displayed at two step: print and wait.
> 
> One change we can think of is to display to the print screen even if there is no printer (without print button).

to be clear, this is pertaining mainly to landscape collages, I realize the portrait already fills the screen vertically.

werdeil, 2020-10-29:
I'm working on a code proposition for this, by using  the "finish" state ("thanks" screen) to show it, but I need to find a way to make it pretty in all cases.

mmk8283, 2021-03-13:
I'm Happy to see there is some progress on this enhancement! Will there be a plugin available? or how do I add this feature? Thanks everyone for your efforts!

mmk8283, 2021-04-21:
I hate to be impatient on this, but my wedding day is rapidly approaching and I would like to get this ironed out as soon as I can. has there been any progress to show the result fullscreen in landscape mode?

werdeil, 2021-04-23:
Hi @mmk8283, unfortunately we don't have much time to work on new feature on pibooth right now. But you can try our dev branch on this topic: https://github.com/pibooth/pibooth/pull/197 and tell us if it fits your need.

anxuae, 2021-05-07:
Available in `pibooth` v2.0.2

Set config key `finish_image_delay` > to 0s to see image in fullscreen before `wait` state.

BR
anxuae

closed 2021-05-07 by anxuae