pibooth/pibooth #105: take picture screen

jo-ei, 2019-12-15:
**Describe the bug**
Hi,
im not sure if its a bug or config problem (or an low Scrren resolution issue).

When taking a picture, the pictograms are in the wrong layer (background)

**Steps to reproduce the bug**
set config landscape orientation resolution 
`resolution = (2592, 1728)`

**Environment**
 - OS : Raspbian Buster
 - Camera: Microsoft Corp. LifeCam VX-2000 OR Canon EOS 1200D
 - Pibooth version: 1.2.0
- Screen resolution: 1280x1024

jo-ei, 2019-12-15:
[image]

werdeil, 2019-12-15:
Thanks for the feedback

It is not really a bug: pibooth is primary developped for portrait photos, and with a portrait setting you will see the pictograms.

I'll try to find a workaround to either put pictograms on foreground or hide them when the preview is in landscape.

jo-ei, 2019-12-15:
hi, maybe the following would be a solution for landscape images.

The distance left, right and above the same size.
The pictograms for it completely downward and into the foreground

(With only one picture, you could also hide the single point)
Please don't hide the pictograms, they're so funny.

werdeil, 2019-12-17:
In fact the pictograms were meant to be on completely downward. To correct this I have changed the way they are loaded on the background.

However I didn't manage to find a way to have them correctly displayed on your screen and preview resolutions. There is not enough space left and I can't find a simple way to put them on top of the preview without reworking our architecture.

Here is what it looks like on a "standard" 16/10 screen with a landscape image
[image]

anxuae, 2020-03-12:
Hi @jo-ei ,

Does the solution described @werdeil fit your expectations?

anxuae, 2020-03-19:
I close this issue, this is normal that in case of wide resolution, pictograms are hidden.

closed 2020-03-19 by anxuae