pibooth/pibooth #106: Hey guys, congrats to this very robust apporach (incl. the ability using pip for updates)

Boeckli, 2019-12-20:
**Describe the bug**
Give a clear and concise description of what the bug is.

**Steps to reproduce the bug**
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error.

**Logs**
1. Set the `debug` mode in the config
2. Post your config file
3. Copy-paste all the logs from the console
4. Post the `diagnostic.log` file if you are using a DSLR camera (generated by running the command `pibooth-diag`)

**Environment**
 - OS [e.g. Raspbian Buster]
 - Camera [e.g. Rpi camera + Nikon D90]
 - Pibooth version [e.g. 1.1.2]

werdeil, 2019-12-22:
Hi, what do you mean by "taylor the pictures"? Do you mean putting a new background?

Boeckli, 2019-12-24:
Hey Vincent, thanks for asking. 
I'm currently seeting up a fresh raspberry OS (buster) in order to see if the problem came from the update 1.1.2 -> 1.2.0. (Incl. --refeshing the .ini file). In 1.1.2 I have had no problem
Before I have used my own pictures for "intro,png, Processing.png, finished.png" (photoshop designed grafics (no standrad text in the forground ("Take picture" and other in the background (pictures on a drying line"). The home made PNG were saved in "pibooth/pictures/DE/(intro,png, Processing.png, finished.png)

werdeil, 2019-12-24:
Ok I think I understand now what happens, since 1.2.0 the texts are now displayed dynamically to allow the change of wording and font. As a consequence I guess that you have your custom image with the text dynamically added on top.

Do you confirm? Can you post a screenshot of what you get?

Boeckli, 2019-12-29:
Vicent, thanks for your emphasis...;-)
I have tried various things now. The problem is still existing and I can state the follwoing:

**My target**: changing the background from plain color to tailored pictures for each of the three status a) Intro 2) processing 3) finished

**My problems**: 
1) Every time, I subtitute the template pictures (arrow.PNG) in the directory ".\home\pi\.local\lib\python3.7\site-packages\pibooth\pictures\assets\arrow.png" the PiBooth software sends an error message "Oops something went wrong" and stops
2) Every time I substitute any a template with a normal picture in 1440x900x24BPP resolution, I see the picture not in color but only in **Black & White**

**My details**: Attached is my one file which causes the problem already "arrow.png". 
Just try it on your installation and copy it in your pi user .local dir.
(I also tried other pictures with 1440x900x24BPP for finished or processing
I'm using PiBooth on Raspi "buster" (Python3.7). I have left all other folder **unchanged** (eg. in usr\local\lib\python3.7\dist-packages\pibooth\pictures\)
Also attached, pease find my "config file" and "translation file". The described OVERLAY problem I have solved it with a "blank" description each
[Pictures_Assets.zip](https://github.com/werdeil/pibooth/files/4008715/Pictures_Assets.zip)

**My Questions**: 
1) Cany you explain, how You can change the the PNG template without causing a problem? (no Ooops and no "Black&White"
2) In the bew version there is no intro.png asset template any more-> How to change the intro picure page ?

Thanks in advance for your help 
Ingo

werdeil, 2020-02-04:
Hi @Boeckli , sorry for the late answer:

> My target: changing the background from plain color to tailored pictures for each of the three status a) Intro 2) processing 3) finished

Is the background the same for all 3 status? Since our rework in 1.2.0 it is not possible to have different background pictures (wasn't meant to be...)
If the background is meant to be the same, you can simply specify the path to the background image in the config.

Arrow and other overlay pictures are now seen as black and white color, then tailored to fit with the background and the text color. So now if you want your arrows to be green (as in the shared picture) just put the text color in green in the config.

About the intro.png no longer present, as there was only text on it and now text is separated from the background we removed it.

I hope it answers your questions. Tell us if you need anything.

Boeckli, 2020-03-03:
Hi @werdeil ,
thanks for the time you have spent on this nice software and esp. on my "problem".

Your last answer explained parts of my problem: You used to have,  but now, you do not foresee anymore three different home made pictures for the three statuses a) Intro 2) processing 3) finished.
I need this personalization option very urgently for my partys, but I can stay with V1.1.2 - right;-)
Solved -> stay with 1.1.2

Then my second problem with V1.2:
It loads the picture but displays it in black&white color only (see Dec 30. and attached files)
Could you duplicate this problem with V1.2?
Thanks

werdeil, 2020-03-03:
As I said for your black&white issue: 

> Arrow and other overlay pictures are now seen as black and white color, then tailored to fit with the background and the text color. So now if you want your arrows to be green (as in the shared picture) just put the text color in green in the config.

So it works as expected, if you want to have a green arrow as in the attached file, keep the arrow file from 1.2 and put the expected color in the config (text color).

Boeckli, 2020-03-15:
Maybe you will find a way in future to have three different home made pictures for the three statuses a) Intro 2) processing 3) finished.
This personalization (possible until V1.2.) was much fun in my parties.
Ingo

closed 2020-03-15 by Boeckli