pibooth/pibooth #107: adjustable resolution of the finished image

jo-ei, 2019-12-23:
**Is your feature request related to a problem? Please describe.**
Hi,
at the moment it looks like the final image is configured with a fixed resolution of 3600 x 2400 (or  2400 x 3600)

**Describe the solution you'd like**
config option for final picture size

werdeil, 2019-12-23:
Can you be more specific do you want other resolution with the same 3/2 ratio or a new ratio?

jo-ei, 2019-12-28:
hi,
i think either the resolution of the monitor is enough to display it fullscreen (#104) or the resolution of the printer (300dpi).

retained 3/2 resolution ?? -> I think yes

or is there an option to configure the used template (with final size, picture size, picture poition, text position, ...)  to create picture process?

anxuae, 2020-03-24:
A solution could be to add a new hook in the plugin system called when the finale picture is created. It will permit to post process the finale picture by changing attributes of the `PictureMaker` or replacing the default one by a  custom `PictureMaker` .

anxuae, 2020-04-01:
Implemented in PR #134

werdeil, 2020-04-06:
As said by @anxuae, since PR #134 you can define your own `PictureFactory` as a pibooth plugin and use the corresponding hooks to do your own template.

I may try to do one as an example soon but in the meantime I'm closing the issue.

closed 2020-04-06 by werdeil