pibooth/pibooth #108: Image deletion (enhancement)

sblanc27, 2020-01-13:
Dear Team,

A new functionality which would be really appreciated is to give the end-user the choice to delete the photo taken for any reason (quality, face, ...), of course a switch on the configuration would be appreciated to activate/deactivate the feature.

A possible implementation of it could be an additional screen between the state "Processing" and "Print".
[image]

werdeil, 2020-01-17:
I don't like adding new interstitial screens, as it increase the workflow time (if the user want to keep the photo).

An option I could think of is to use the second button on the print page, we would have something like "<- please forget this ugly photo" "Print the photo ->"

But this won't address the usage without printing.

@anxae what do you think?

anxuae, 2020-02-03:
I agree with you @werdeil, a new arrow can be added below the picture to print in order to "forget" the picture

anxuae, 2020-03-24:
@werdeil should we make the "print" state always displayed since it is used to propose the image deletion?
(Currently the state is skipped if no printer installed)

werdeil, 2020-03-24:
For me not as it is now, if we make it always displayed we shall change the complete layout (including arrows) depending on the fact that there is print or not.

I see the forgot feature most as a way to avoid waiting time on the print screen, if there is no print option user doesn't wait so no need to add an action here...

anxuae, 2020-03-24:
That's ok for me.

Other point: the folders in `raw` should not be also moved in a raw folder inside the `forget` one?

werdeil, 2020-03-24:
I would say no too, for me all the raws shall be kept in the same folder.

anxuae, 2020-03-24:
But a least a folder `forget` in the `raw` one, be cause if I want to upload in google drive all the raw photo, I need to select manually which one is forgotten or not...

werdeil, 2020-03-24:
I disagree on that, for me raws shall be kept as they are.

There can be good raw images from a forgotten final picture, I don't think we shall put them away.

closed 2020-04-06 by werdeil