pibooth/pibooth #109: terms and conditions screen (enhancement)

sblanc27, 2020-01-13:
Dear Team,

I know, this functionality does not sound really funny, but because of the GDPR laws a formal Opt-In might be required in such case and also an explanation of the usage made with the data.

A possible implementation of it could be an additional screen between the initial state the Choose Layout state.

Of course, having the possibility to change the text and title on the configuration file as well as activate the feature would be required.

[image]

werdeil, 2020-01-17:
In my opinion, your possible implementation won't address the GDPR needs, if we want to store a consent we need a way to clearly identify the person using the pibooth (enter a name / mail) and log it somewhere.

Moreover from what I understood while looking into the GDPR laws, in a case of a private usage (as the pibooth intended use) the GDPR laws is not applicable. If the pibooth is used in a professional context, it is on the pibooth's main user responsibility to respect the GDPR laws and as actual capacities of the pibooth project doesn't allow the user to give an identifier this has to be done outside of the pibooth code.

So I would say that to solve this issue the only thing we can do on our side is to add GDPR advices for our users in the README section of the project.

werdeil, 2020-02-02:
I've added some advice on the README page in the commit e52a015892d13d7e52a70ddbbd46aefe0acf7ae0, closing this issue as a won't do.

closed 2020-02-02 by werdeil