pibooth/pibooth #110: Pi Zero or Pi 1. Generation

pedakas, 2020-01-25:
Hello, 
I have a short question, will it work on a RPI Zero or a RPI 1. Generation?

Thanks!

werdeil, 2020-01-30:
Technically it should work, but it will be too slow to be useable I think.

We didn't test on our side with a RPi zero nor Roi 1 but when we tested on a RPi 2 it was too slow. That's why we tell "1 Raspberry Pi 3 Model B (or higher)" in the README.

pedakas, 2020-01-30:
Thank you very much!

closed 2020-01-30 by pedakas

mattdsteele, 2021-03-24:
Hi folks, I have a similar, but opposite question. Has anyone tested on the Raspberry Pi 4? Any idea whether you would see significant performance improvements?

Since my current hardware (Pi Zero) won't cut it, I'm wondering if it's worth spending a little more buying the latest 4 model.

werdeil, 2021-05-28:
Hi @mattdsteele Sorry for the late answer as the issue is closed we didn't see your question.

I would say the bigger the better, appart from the global speed, the main slowness we get on old Raspberry Pi is processing part (where we merge the capture in one picture), the last Raspberry Pi will be better on this part.

mattdsteele, 2021-05-28:
No worries, I ended up picking up a 4 in the interim, and it's been working well!

jeyremy, 2022-09-20:
@mattdsteele May i ask what version of the 4 are you using ? 
i'm about to buy one... but i dont know if 2gb will be enought or go to the 4gb version...

mattdsteele, 2022-09-21:
@jeyremy I used the 2gb version, and it seemed to work fine!