pibooth/pibooth #111: Settings menu opens when hitting one of the buttons

Exitras, 2020-01-26:
Hey, first of all, many thanks for this very nice project! 

Unfortunately, I experienced a very annoying bug which makes it unusable for me in the current version.

**Describe the bug**
Sometimes, when I press one of the two keys (I don't think it matters which one), the settings menu opens. When I press the key again, the selected option in the menu goes up or down but I can not close the settings menu again without hitting the ESC key on a keyboard.
It doesn't matter whether an external keyboard or mouse is connected. It happens either way. 

**Steps to reproduce the bug**
It happens on the main screen (where we are asked to take a photo) and on the printing screen.

**Logs**
I don't have a log yet but I can create one if you are interested.

**Environment**
 -  Raspbian Buster
 -  Raspberry Pi 4
 - Canon 2000D
 - Pibooth version 1.2.0
 - I am using 2 buttons but no LEDs or resistors.

werdeil, 2020-01-30:
Hi, I think this is linked to the debounce delay of your buttons.

Have a look at the what have been answered here: https://github.com/werdeil/pibooth/issues/83#issuecomment-532300766

Tell us if it helps.

anxuae, 2020-03-03:
Hi @Exitras ,

Have you try the solution proposed by @werdeil ?

closed 2020-03-12 by anxuae