pibooth/pibooth #112: More of a question.

DaddioD, 2020-01-29:
How did you get your landscape final to have that small of margins. All my attempts seem to make the picture smaller.

[image]

werdeil, 2020-01-30:
Hi DaddioD,

I think that you are looking for the "capture_cropping" setting, which is by default set to false. Have you tried with this setting set to True?

You can either change it from the file in the .config folder or via the settings menu, in the picture submenu, item "Crop captures".

Tell us if it answers your question.

DaddioD, 2020-02-02:
Sorry to respond late. Will be able to try that tomorrow.

anxuae, 2020-03-03:
Hi @DaddioD , 

Have you try the solution proposed by @werdeil ?

anxuae, 2020-03-19:
I close this issue, because I have no news.
Please reopen it if issue is persisting.

closed 2020-03-19 by anxuae