pibooth/pibooth #113: Change BG Images / Update to 1.2, help needed

dvid9976, 2020-02-02:
hello there! first of all, thank you very much for such a great proyect and hard work.
Im building pibooth for my daughter's birthday party and this your project has something that many others has not.. a huge amount of time putting together clear instructions on how to get it to work :-)

I do have pibooth already working but strugle to change bg images, overlays, etc; how can I know what version it is? how can I change bg images?

I realized that all that stuff is available in V1.2 but it I don't know how to upgrade :-(

I downloaded the packages hoping that I could find a procedure into it, but still no way.
Any help pls?

thank you very much in advance

werdeil, 2020-02-02:
You can check your `pibooth` version by typing the command `pibooth --version` on a terminal.

If you have installed `pibooth` through pip you can simply upgrade via the command
`pip install --upgrade pibooth`

Once you are up to date, you need to open the config file to do your changes:
- the background image is in the `WINDOW/background` field, just put the path to the image you wan to use.
- for the overlay it is on `PICTURE/overlays`, and it accepts a list of overlays if you want a different one for the 1 picture and 4 pictures.

Tell us if it solve your questions, or if you need more help.

dvid9976, 2020-02-04:
hello mate!
sorry for the late reply; sure you solved my question, thank you very much for that.

I have a last one.. are the png (choose a picture, choose layout, processing.. etc) also customizable? I have been able to find the gimp files but as far as I don't use Gimp, don't know if it si so simple as replace the existing ones by new ones.

than you in advance!

werdeil, 2020-02-04:
As I explained in another issue, since version 1.2.0 there is no longer dedicated image for each step but only overlays (such as arrow.png, processing.png, layout2.png...). These overlays are considered as a white drawing on a transparent background (to be put over the background and colorized according to the text color).
You can change any of these overlays to one of your choosing if you want, just keep in mind that they are considered as white drawing on transparent background by pibooth

dvid9976, 2020-02-04:
thank you sir!

closed 2020-02-04 by dvid9976