pibooth/pibooth #115: Introduce a plugin system

werdeil, 2020-02-04:
Introduce a plugin system (eg https://pypi.org/project/pluggy ) to let the possibility to integrate custom functions at specific steps of the pibooth process.

It will allow the integration of dedicated behavior outside of the main pibooth code, for example create a plugin to:
- Publish picture at the print step #52
- Neopixel support at multiple steps #100 
- Extract printing out of pibooth as it is not a mandatory step
...

This is a big feature, setting a milestone for pibooth 2.0

anxuae, 2020-03-12:
First implementation in https://github.com/werdeil/pibooth/tree/dev-pluggy

anxuae, 2020-03-21:
PR under review https://github.com/werdeil/pibooth/pull/124

closed 2020-03-24 by anxuae