pibooth/pibooth #116: photo rotation

oksimple, 2020-03-02:
Hi friend, your project is awesome, but I think if you can rotate the photo, it will be more widely used.

[image]

anxuae, 2020-03-12:
This proposal is interesting, however it is far to the current implementation of the ``pibooth`` software.
It require lots of picture handling thanks to PIL or OpenCv libs, and i think, the orientation of each picture need to be defined in the configuration, which may be tricky for the newbie.

oksimple, 2020-03-14:
@anxuae I think it can be saved to the situation of the template. Select a template before taking a picture and replace the placeholder picture in the template after taking a picture. This is not complicated for novices and can make the template widely shared.

anxuae, 2020-03-24:
A solution could be to add a new hook in the plugin system called when the finale picture is created. It will permit to post process the finale picture by changing attributes of the `PictureMaker` or replacing the default one by a  custom `PictureMaker` .

werdeil, 2020-04-06:
As said by @anxuae, since PR #134 you can define your own `PictureFactory` as a pibooth plugin and use the corresponding hooks to do your own template.

I may try to do one as an example soon but in the meantime I'm closing the issue.

closed 2020-04-06 by werdeil