pibooth/pibooth #117: add external commands

gliepins, 2020-03-06:
Dear friends,
It may very well not be a feature request, however this is what I want to ask:
1) I want to add a simple shell script that runs every time after photos have been taken. I want the script to upload the resulted photos to a separate web server. 
Please, tell me where and if I can "add" such script? 

Thank you.

anxuae, 2020-03-12:
Hi @gliepins ,

this feature is on the top of ours priorities. This will be available when #115 will be implemented.

BR
anxuae

anxuae, 2020-03-18:
Hi @gliepins ,

Can you please provide me an exemple of such script written in python to upload photo?
I would like to do some tests with the ``pibooth`` plugin system.

thanks
anxuae

gliepins, 2020-03-18:
Hi. 
In short - i now run a simple "qrencode" command to generate qr code that links the photo with webserver for a quick photo download. 
It would be helpful for such command to be executed automatically. 
And also I want to display/show that qr code besides the photo on the left. 

/G

anxuae, 2020-03-24:
Will be possible in next `pibooth` major release (date to be planned).
See plugins example at https://github.com/werdeil/pibooth/blob/master/docs/plugin.rst .

closed 2020-03-24 by anxuae