pibooth/pibooth #118: Flash

jasharrar, 2020-03-13:
Sorry of this has been asked before but with the LED GPIO is there a way for the led to flash only when the picture is taken rather than for the whole count down?

thanks in advance

anxuae, 2020-03-16:
Hi @jasharrar ,

This was a request from @thifi in #25 , thus I don't want to change this behavior.
An other the solution could be to drive a new GPIO during flash.

@werdeil an idea?

jasharrar, 2020-03-16:
Thanks, it would be good to add this as a new gpio option. This gives more flexibility to the end user. Thanks again

werdeil, 2020-03-17:
As @anxuae I don't want to change the behavior of current GPIO, and I don't want to add new GPIO on the core code. 
I would say we will wait on the work on #115 to have a plugin for GPIO handling during capture.

jasharrar, 2020-03-17:
Totally appreciate that, thanks guys. Do we have a time scale for 115? Depending on Corona virus I'm getting married in August and it would be good for then. I'm working on a new build for my booth over the next few weeks and just got a dnp620 printer. I'll share some pics when it's finished. Thanks again

anxuae, 2020-03-19:
Hi @jasharrar,

Yes, I have submitted a PR on this topic, it is almost finished (I have time because France is completely confined due to covid19). 
I hope for you that your wedding will be hold. Mine was scheduled in 2 days and was cancelled...

BR
anxuae

anxuae, 2020-03-24:
Will be possible in next `pibooth` major release (date to be planned).
See plugins example at https://github.com/werdeil/pibooth/blob/master/docs/plugin.rst .

closed 2020-03-24 by anxuae