pibooth/pibooth #119: Background template

TheSuperpapi, 2020-03-14:
Hello,

Is there a way to change the final background template in order to have a costomize background instead of White ?

TheSuperpapi, 2020-03-14:
Bonjour,
Y a t'il un moyen de changer le template de fond sur la photo finale pour passe de l'arrière plan blanc à une image de fond personnalisée ?

werdeil, 2020-03-14:
Hi, there is a way to change the picture background through the config file with the key "backgrounds" of the PICTURE part.
Default value is white (255, 255, 255), but you can put any RGB color, or a path to a file.
You can see example of what is possible in the [gallery](https://github.com/werdeil/pibooth/blob/master/docs/examples.rst)

Tell me if this answer your question, I let the issue open as I think some info can be added in the doc to help users find this feature.

enricofusco, 2020-03-15:
I don't understand how to change the size of the photo in the preview of the shot and have it large original horizontal.

Output file
how to remove the white border and have the large photo
and add a watermark

TheSuperpapi, 2020-03-15:
Hello,

This is exactly what i want to de, i'll try is as soon as i have à quarter 😉

werdeil, 2020-03-15:
@enricofusco I have opened a new issue for your question to avoid spamming this one.

TheSuperpapi, 2020-03-15:
Big Thanks

closed 2020-03-15 by TheSuperpapi