pibooth/pibooth #120: Change the size of the photo in the preview of the shot and have it large original horizontal.

werdeil, 2020-03-15:
I don't understand how to change the size of the photo in the preview of the shot and have it large original horizontal.

Output file
how to remove the white border and have the large photo
and add a watermark

_Originally posted by @enricofusco in https://github.com/werdeil/pibooth/issues/119#issuecomment-599156813_

werdeil, 2020-03-15:
Hi @enricofusco, not sure to understand your questions but here are some answer:

> I don't understand how to change the size of the photo in the preview of the shot and have it large original horizontal.

You have to change the `[CAMERA] resolution` in the config, default value is `(1934, 2464)`, you have to find a new one compatible with your camera.

> Output file
> how to remove the white border and have the large photo

There is no way to remove the border, you can change the color, have an image as background but for now it is not possible to tweak the output in a way to not have a border (but you can open an issue to add this kind of option, we will see if we find a way to do it).

> and add a watermark

To add a watermark you need to use the overlay feature in the config `[PICTURE] overlays`. You need to prepare a PNG with your watermark (and transparency) and then put the path to this PNG in the config.

Tell me if it answer your questions.

enricofusco, 2020-03-15:
I don’t understand how to do it😅 ... 
I tried but it doesn't work

anxuae, 2020-03-16:
Hi @enricofusco 
Can you be more explicite please? What have you tried?

Note that the configuration shall be edited using the command: `pibooth --config`

anxuae

enricofusco, 2020-03-16:
if i use the comand pibooth --config, i can't edit it....

anxuae, 2020-03-16:
What is the error please?

enricofusco, 2020-03-16:
I can't write or save

anxuae, 2020-03-16:
Can you run the command `ls -la ~/.config/pibooth/` and give me the output ?

enricofusco, 2020-03-16:
-- 

*Enrico Fusco*
*Digital content creator*
Facebook <https://www.facebook.com/enricofuscophoto/> - Instagram
<https://www.instagram.com/enricofusco/>
www.enricofusco.it
_

*Questo messaggio e gli eventuali allegati sono ad uso esclusivo del
destinatario e possono contenere informazioni riservate e/o confidenziali
la cui riservatezza è tutelata legalmente. Nel caso in cui chi riceve il
messaggio non sia il destinatario designato, chiediamo di non farne copia,
non diffondere, non usare, non divulgarne il contenuto. Chiunque riceva
questo messaggio per errore è pregato di distruggerlo e di segnalarci
l'errata ricezione.Please advise immediately if you or your employer do not
consent to Internet email for messages of this kind. Opinions, conclusions
and other information in this message that do not relate to the official
business of our firm shall be understood as neither given nor endorsed by
it. We take reasonable precautions to ensure our Emails are virus free.
However, we cannot accept responsibility for any virus transmitted by us
and recommend that you subject any incoming Email to your own virus
checking procedures.*

enricofusco, 2020-03-16:
[image]

anxuae, 2020-03-16:
The configuration file has been opened several times, without being closed properly. This may be the reason why you can save the configuration anymore (I don't how it happened).

Best is to remove the all the configuration before starting `pibooth` again.
Execute this command: `rm -rf ~/.config/pibooth/`

anxuae, 2020-03-19:
Hi @enricofusco 

Does it solved your problem?

BR
anxuae

enricofusco, 2020-03-19:
yes I understand how to edit.
what I'm asking now is: 
1) will you ever add the possibility of choosing whether to have:
     A) the borderless photo with only the logo
     B) the photo as it is now

2) choose to enable printing or not in the screens

3) understand how to have a button to take photos ,like the one in the photo below (without cables)

[image]

anxuae, 2020-03-20:
> will you ever add the possibility of choosing whether to have:
> A) the borderless photo with only the logo

Borderless is not possible at the moment
Logo shall be added using `overlay` feature (see configuration)

> B) the photo as it is now

Not possible and not planned

> choose to enable printing or not in the screens

To hide the print view, just set ``[PRINTER][printer_delay]`` to 0

> a slideshow in a second monitor with all the photos taken

You have to find an other software to do that

> understand how to have a button to take photos ,like the one in the photo below (without cables)

Look at the circuit diagram https://github.com/werdeil/pibooth#circuit-diagram

enricofusco, 2020-03-21:
> > understand how to have a button to take photos ,like the one in the photo below (without cables)
> 
> Look at the circuit diagram https://github.com/werdeil/pibooth#circuit-diagram

I'm not familiar with these things,
I don't know what button to take and how to connect it in this way

[image]

werdeil, 2020-04-01:
> I don't know what button to take

You are looking for a "massive arcade button" like this one: https://www.adafruit.com/product/1185

But we are getting out of the original topic, do you still have issue with the preview or can we close this topic?

anxuae, 2020-04-09:
I close this issue.
Feel free to open a new one if necessary.

closed 2020-04-09 by anxuae