pibooth/pibooth #122: change color text if use background image

sravel, 2020-03-18:
Thank you for this great code which corresponds exactly to what I am looking for.

However, I have a small modification that I would like.
- My problem is that I have a mid image in the background and the majority color which is light green. but on the white background of the photos it is not visible. By looking at the code I managed to modify in the file "background.py" line 148
```self._background_color = pictures.get_pygame_main_color(self._background)```
to
```self._background_color = (0,0,0)  #tuple color```

- Another question: is it possible to not recolor the images in black and white?
see lines 71/72 of the pictures / __ init __ () file optional ?

**Describe the solution you'd like**
- One solution would be to set image and / or text color
example: if image alone, majority color
otherwise image and text color

- solution for 2
```
if color and to_black_and_white: #to_black_and_white is boolean load on config file
    pil_image = set_picto_color(pil_image, (255,255,255), (0,0,0))
```
[image]

Thanks

Sébastien

werdeil, 2020-03-18:
Hi thanks for the feedback

I'm not sure I understand: your issue is that the text is in light green as your background main color is in light green too?

sravel, 2020-03-18:
yes and no, i want to change the color of text "1 photo" and "2 photos", because i use background image and your code select the main color of image background. 
if I change the code line
```self._background_color = pictures.get_pygame_main_color(self._background)```
to 
```self._background_color = (0,0,0)```
the text color change in black
it is better to have parameter in the file config for adjust color instead of change the code.

werdeil, 2020-03-18:
> your code select the main color of image background.

Yes this is the intended behavior so the whole thing work smoothly with any background/ text color

Have you considered changing the text color? it light-green on white is not readable so is white on light-green

Here is a small test I just did with a light-green background and white or black text color.

[image]
[image]

sravel, 2020-03-18:
Ok I understood, in fact my two points are linked.
I will try to explain my request better with examples of capture:
in the config file, I have the color of text in white
and an image in the background.
if i go back to the original code when i launch i have this rendering:
[image]

Now I would like to keep the image in color (the butterfly) so I comment the code on file pictures / __ init __ () line 71
```
if color:
             pil_image = set_picto_color (pil_image, color, bg_color)
```
i have this rendering:
[image]
so now I want to change only the color of the text "1 photo" and "2 photos" to RGB color define on config file.
for the moment i change the code on file  "background.py" line 148
to ```self._background_color = (0,0,0) #tuple color```
so I have this rendering:
[image]

the aim of my issue is to have somme option on config file instead of change code.
I now python so I try to modify but i don't explore all your code for the moment !!!!

Thanks

closed 2020-03-26 by sravel