pibooth/pibooth #125: Multiple folders save

anxuae, 2020-03-21:
**Is your feature request related to a problem? Please describe.**
Save pictures in 2 devices in case of SDCard corruption

**Describe the solution you'd like**
Can define several paths in the configuration key ``[GENERAL][directory]``

- All folders have to be cleared if ``clear_on_startup=True``

anxuae, 2020-03-21:
@werdeil this feature has been asked in #42 by @moskate49 and @Aurel-Charles.
I create an issue to discuss the interest of it.

werdeil, 2020-03-22:
I don't really see the interest, as we already can save on an external usb is needed.

But if it is easy to implement why not.

Rather out of subject but I'm considering removing the `clear_on_startup£ option as it is very dangerous and only useful for debug purpose.

anxuae, 2020-03-22:
Ok so this issue can be closed.

Otherwise, for the `clear_on_startup `, we can add a guard in order to enable it only in debug mode (this option is really useful during development).

werdeil, 2020-03-22:
> this option is really useful during development

Never used it :-)

moskate49, 2020-03-22:
In my configuration the pi is enclosed in a box who is cant be opened. The usb key is outside the box but during a party a drunk boy touch the key and break it :/ .
Multiple folder in this case is usefull for have a « B plan ».
The externe key is usefull to avoid multiple opening and close the box at every party . But internal save is in case of .
Thanks for your work
(Sorry for my bad english)

anxuae, 2020-03-23:
The effort is not so huge, I think it can be easily implemented.

anxuae, 2020-05-16:
Implementation to be tested in the PR #146

Exitras, 2020-08-11:
Hi, I am rather new to Raspberry Pi but want to use Pibooth for my photo booth and so far, everything works just fine. However, I would like to store the images on the internal SD Card and an external USB Stick to avoid data loss in case of some SD Card failures. I don't understand how to use the "multiple folders save" function. I tried "directory = ('~/Pictures/pibooth', 'media/pi/PIBOOTH/images') with no luck. 
Furthermore, it would be rather beneficial if the RAW images could be stored in one folder without the subfolders. Is there a way to do this? This would only be needed for the USB Stick.

anxuae, 2020-08-11:
Hi @Exitras 

Did you have an error when you changed the configuration?

Can you try with a `/` in front of `media`?
`directory = ('~/Pictures/pibooth', '/media/pi/PIBOOTH/images')`

Please post the logs generated during a sequence of captures (start `pibooth` with `-v` option)

Unfortunately, it is not foreseen to manage 2 locations with 2 different directories structures

Exitras, 2020-08-11:
Hi Anxuae, many thanks for your quick reply. 
I tried your suggestion but it didn't work. The pictures are now stored in "/home/pi/.config/pibooth/('/media/pi/PIBOOTH/Bilder','~Pictures/pibooth')"
It actually created new folders such as (' and so on. Do I have to activate something in order to create two directories?

Exitras, 2020-08-11:
Hi again :) 
I solved it by using a cronejob workaround that syncs all images every 5 minutes.
*/5 * * * * rsync -a /home/pi/Pictures/pibooth/raw /media/pi/PIBOOTH/Bilder
Not the best solution but it works for now.

anxuae, 2020-08-12:
Hi @Exitras ,

I need your configuration file and your logs to investigate. Without that, can't do anything.

Exitras, 2020-08-18:
It works now :) Thanks for your support!
(Don't forget to upgrade to Pibooth 2.0 before)

commit c97a5f42 (2020-05-22): Merge pull request #146 from pibooth/multi-folders-save

Multi folders save to fix #125
closed 2020-08-18 by anxuae