pibooth/pibooth #127: Extract startup_led et preview_led in an external plugin

werdeil, 2020-03-22:
As we are introducing a new plugin system and as these 2 leds came from an external request and are not for me part of the core pibooth we shall remove them from pibooth and intergrate them in a dedicated plugin.

Note: I'm not sure it is easy for the startup_led.

anxuae, 2020-03-24:
2 new hooks can be added in order to manage actions at `pibooth` startup and cleanup to manage to `startup_led `.

sravel, 2020-03-31:
hello, I test use the __gpiozero__ lib (#131) and for the initialization you can use this : 
```self.led_startup = LED("BOARD" + config.get('CONTROLS', 'startup_led_pin'), initial_value=True)```

I implemented RGB led as plugin with gpiozero lib and i actually make some test. I can test add core led as plugin if you want.

anxuae, 2020-05-05:
I propose to create a repository named `pibooth-lights` to move the extra lights in.
What do you think about this name?

werdeil, 2020-05-06:
I would say pibooth-extra-lights to emphasze the fact that « standard » lights are not in this one.

anxuae, 2020-05-06:
It seems great to me!

anxuae, 2020-05-10:
LEDs are extracted.
First version of the plugin is now available on PyPi: https://pypi.org/project/pibooth-extra-lights
This issue can be closed

closed 2020-05-11 by werdeil