pibooth/pibooth #128: Add RGB config element available in config menu

werdeil, 2020-03-22:
**Is your feature request related to a problem? Please describe.**
As config is big and the choice of text, background for display and final picture is quite important I think we shall be able to set them through the config menu

**Describe the solution you'd like**
Ideal: writing separately Red, Green and Blue value and have a preview square where the value is computed to show the color
Acceptable: a text setting where we put (R,G,B) directly.

**Describe alternatives you've considered**
A separated tool to generate the config at start

anxuae, 2020-03-22:
I propose to ask to @ppizarror if such a widget can be added to pygame-menu

ppizarror, 2020-03-23:
hmmm, such widget may be 3 different text boxes and a preview color. Something like this:
[image]

anxuae, 2020-03-23:
Hi @ppizarror, thanks for your quick comment ;-)
Yes this seems a good idea. To a void a too long widget, we could use the background of entry text to put the color:

[image]

(managing font color in white when color is too dark)

werdeil, 2020-03-23:
I prefer the first option, having the background changr is quite strange no?

werdeil, 2020-03-23:
However if we have only one textview with the 3 values I find it more understandable (as it is updated once I have changed the 3 of them).

Color [ (23, 54, 123) ]

ppizarror, 2020-03-23:
> However if we have only one textview with the 3 values I find it more understandable (as it is updated once I have changed the 3 of them).
> 
> Color [ (23, 54, 123) ]

but the user must enter a comma for each number, that's not so user-friendly. I think that 3 boxes with each number separated is much easier to use

werdeil, 2020-04-01:
> user must enter a comma for each number

Yes indeed, this doesn't seems to be an issue for me as my first idea was to have something like on this website: https://www.w3schools.com/colors/colors_rgb.asp
User can specify the color either in RGB or in hex 

Or we keep only the RGB option and in this case let only the number editable (I don't know if you have already something for this in pygame-menu)

ppizarror, 2020-04-04:
New version v2.2.0 online! This introduces color input widget.

PR #136 (2020-04-07): Integrate pygame-menu 2.2.0
Integration of pygame-menu 2.2.0 and first use in window/text_color config

Fix #128
closed 2020-04-07 by werdeil