pibooth/pibooth #129: Previous capture not displayed when current is forgotten

anxuae, 2020-03-23:
**Describe the bug**
When a picture is forgotten, the previous one is no more displayed on the "wait" state.

**Steps to reproduce the bug**
1. Run a first capture sequence
2. Run a second capture sequence
3. Forget the picture of the second sequence
4. The first capture is not displayed on the "wait" screen

**Environment**
 - Mac OSX
 - Camera embedded
 - Pibooth version 1.2.0 dev

werdeil, 2020-03-23:
For me it is the expected behavior, not sure that displaying an old image would be understandable...

werdeil, 2020-03-23:
What about adding a way to define a default image that will be shown at startup and when the image is forgotten?

anxuae, 2020-03-23:
Yes this can be a possibility.

However viewing the photo of friends during a party is funny, and for me there is no ambiguity to show the last not forgotten picture.

anxuae, 2020-03-24:
Proposal 1

[image]

anxuae, 2020-03-24:
Proposal 2

[image]

https://www.pinterest.fr/pin/791085490779175668/

werdeil, 2020-03-24:
But I thought you wanted the previous unforgotten picture?

anxuae, 2020-03-24:
A nice picture is also ok for me ;-)

werdeil, 2020-03-24:
Need to handle the print option: what do you think: we gave the option to print the second previous or we show the image but not the option to print it?

anxuae, 2020-03-24:
Maybe don't need to print, the second previous may already be printed several times.
A simple solution is set the `app.nbr_duplicates` to a high value.

werdeil, 2020-03-26:
Added the app.nr_duplicates part too, please review :-)

werdeil, 2020-03-31:
Solved by https://github.com/werdeil/pibooth/commit/4c2bc2a6874d19907eb3c1c6e1246fefda85317a

closed 2020-03-31 by werdeil