pibooth/pibooth #131: add python gpiozero lib

sravel, 2020-03-26:
just a good lib for use all gpio pin
https://gpiozero.readthedocs.io/en/stable/index.html
I think you can use it for button and led control

https://gpiozero.readthedocs.io/en/stable/index.html

anxuae, 2020-03-31:
Hello @sravel ,

I know this library, however the choice has been made to use RPI.GPIO. We have classes to manage buttons and leds in `pibooth`. I don't really understand the interest of adding this dependency...

anxuae

sravel, 2020-03-31:
with the plugin system it will be easier to include other system: digital number matrix, RGB Led, ...
it also simplifies code management
for example:
```
if event.type == BUTTON_DOWN:
                 if event.pin == self.button_capture and (type_filter is None or type_filter == event.type):
```
become
```
if self.button_capture.is_pressed:
```
it is also easier to make false GPIO

```
try:
     from RPi import GPIO
except ImportError:
     print ("toto")
     # GPIO library only existes on Raspberry Pi
     # Set the default pin factory to a mock factory
     from gpiozero.pins.mock import MockFactory, MockPWMPin
     from gpiozero import Device
     Device.pin_factory = MockFactory (pin_class = MockPWMPin,)
```

anxuae, 2020-03-31:
Yes I understand the "simplicity" of this lib.
It imply lots of changes in the `pibooth` core software.

werdeil, 2020-03-31:
Maybe we can either make pibooth setup the board as BCM and use the corresponding GPIO I/O:
(example 11 --> 17, 7 --> 4, ...)?

Or add a setting for the GPIO setup, but you know that I don't like adding settings but here it can be acceptable.

Would this change make the use of both libs possible?

anxuae, 2020-03-31:
As `gpiozero` can be used in BOARD mode, I recommend to keep BOARD config which is compatible with all RPI models.

sravel, 2020-03-31:
Default is BCM but you can use BOARD or other like this:
```
>>> led = LED(17)
>>> led = LED("GPIO17")
>>> led = LED("BCM17")
>>> led = LED("BOARD11")
>>> led = LED("WPI0")
>>> led = LED("J8:11")```
I think like you, keep BOARD is better. 
@werdeil  I think it's possible to change BOARD to BCM with dict to keep user config with BOARD.
because to work with both lib I change
```       # Prepare GPIO, physical pins mode
        GPIO.setmode(GPIO.BOARD)
```

werdeil, 2020-03-31:
Ok then there are no issue, anyone can use gpiozero lib in an external plugin?

sravel, 2020-03-31:
I started to change GPIO to use gpiozero, you can found here
https://github.com/sravel/pibooth/tree/gpiozero

sravel, 2020-03-31:
@anxuae 
> It implies lots of changes in the `pibooth` core software.
I don't understand all of your code yet, but what changes need to be made? and where?

anxuae, 2020-03-31:
Yes I know.
I saw your changes, it is a good start. However some features might broken due to your changes.
For instance: synchro blinking lights, filtering events depending the HW source is no more functional in your changes

sravel, 2020-03-31:
OK I restore filtering events.
For led Synchor I can use this
https://gpiozero.readthedocs.io/en/stable/recipes.html#ledboard
But I don't understand which led blink syncho. Thanks.
Other question is it possible to have some methode to call with you directly? If you want.

sravel, 2020-04-01:
I export all lights on plugin, and remove from app, If you can try

werdeil, 2020-04-01:
Led sync is implemented each time the leds can blink together, so each time the both buttons are available, for example during layout choice. At first we didn't have that and it was very strange. The ledboard option can be a solution indeed.

I appreciate your work on this topic but I still don't understand why you want to add a new third party lib to `pibooth`. Yes this lib is interesting and easy to use, yes it can be interesting for implementing new leds/button but none is planned. So if you manage to integrate it and send us a PR we may consider changing.

About speaking with us directly, you can find our mails in the git history, and as we are both french speaking you can talk directly in french :-)

werdeil, 2020-04-14:
Implemented in #137, thanks @sravel

closed 2020-04-14 by werdeil