pibooth/pibooth #142: Developping instruction installation not working

TheSuperpapi, 2020-05-12:
Hello, I wanted to try curent dev version but seems not working when following te dev instruction.
I've follow the normal installation since the point 7 were I switch to the github clone as mentionned.

Thanks for the help

Returned message:

pi@raspberrypi:~/pibooth $ sudo pip3 install -e .[dslr,printer]
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Obtaining file:///home/pi/pibooth
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/home/pi/pibooth/setup.py", line 12, in <module>
        import pibooth
      File "/home/pi/pibooth/pibooth/__init__.py", line 7, in <module>
        import pluggy
    ModuleNotFoundError: No module named 'pluggy'
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /home/pi/pibooth/

anxuae, 2020-05-12:
Hi @TheSuperpapi 

Can you try again? I've just pushed a fix.

TheSuperpapi, 2020-05-12:
pi@raspberrypi:~/pibooth $ sudo pip3 install -e .[dslr,printer]
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Obtaining file:///home/pi/pibooth
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/home/pi/pibooth/setup.py", line 12, in <module>
        import pibooth
      File "/home/pi/pibooth/pibooth/__init__.py", line 7, in <module>
        import pluggy
    ModuleNotFoundError: No module named 'pluggy'
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /home/pi/pibooth/

TheSuperpapi, 2020-05-12:
still the same :(

anxuae, 2020-05-12:
Can't reproduce on my side, do you pull the changes?

TheSuperpapi, 2020-05-12:
Sorry, i missed tu pull :-( it works fine now.
Big thanks

TheSuperpapi, 2020-05-12:
I now have this error when lauching pibooth.Strange, will check again.
....
File "/home/pi/pibooth/pibooth/config/menu.py", line 7, in <module>
    import pygameMenu as pgm
ModuleNotFoundError: No module named 'pygameMenu'

anxuae, 2020-05-12:
It's "normal" because it's a developing version ;-)

We are integrating a new version of the menu, temporary fix the error by running:

```
pip install pygame-menu==2.3.3
```

TheSuperpapi, 2020-05-12:
Haaa hats why, big thanks :)

TheSuperpapi, 2020-05-12:
Hi again,

pip install pygame-menu==2.3.3 return:

Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: pygame-menu==2.3.3 in /usr/local/lib/python2.7/dist-packages (2.3.3)
Requirement already satisfied: pyperclip in /usr/local/lib/python2.7/dist-packages (from pygame-menu==2.3.3) (1.8.0)
Requirement already satisfied: pygame>=1.9.4 in /usr/lib/python2.7/dist-packages (from pygame-menu==2.3.3) (1.9.4.post1)

But pibooth return:
pygame 1.9.4.post1
Hello from the pygame community. https://www.pygame.org/contribute.html
Traceback (most recent call last):
  File "/usr/local/bin/pibooth", line 11, in <module>
    load_entry_point('pibooth', 'console_scripts', 'pibooth')()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 489, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2793, in load_entry_point
    return ep.load()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2411, in load
    return self.resolve()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2417, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/home/pi/pibooth/pibooth/booth.py", line 27, in <module>
    from pibooth.config import PiConfigParser, PiConfigMenu
  File "/home/pi/pibooth/pibooth/config/__init__.py", line 4, in <module>
    from pibooth.config.menu import PiConfigMenu
  File "/home/pi/pibooth/pibooth/config/menu.py", line 7, in <module>
    import pygameMenu as pgm
ModuleNotFoundError: No module named 'pygameMenu'

anxuae, 2020-05-12:
There is a mismatch in the python version:

You have installed `pibooth` with python3
```
File "/usr/lib/python3/dist-packages/pkg_resources/init.py", line 489, in load_entry_point
```

But you have downgraded `pygame-menu` in your python2 distribution:
```
Requirement already satisfied: pygame-menu==2.3.3 in /usr/local/lib/python2.7/dist-packages (2.3.3)
```

Please run:
```
pip3 install pygame-menu==2.3.3
```

TheSuperpapi, 2020-05-12:
Yes it works fine now, I did not see that......
Sorry and thanks again for your support :)

Antoine125ys, 2020-05-17:
Hi, I have the same error with pygame-menu. I'm a beginner, and I don't quite understand how to install python 2 and 3. By retrograding,... Can you help me? I'm despairing.

anxuae, 2020-05-19:
Hi @Antoine125ys,

First of all, can you tell me how did you installed `pibooth`?

TheSuperpapi, 2020-05-19:
Hello, here the process list of command line to enter (from the beginning).
$ sudo apt-get update
$ sudo apt-get upgrade
$ sudo wget raw.github.com/gonzalo/gphoto2-updater/master/gphoto2-updater.sh
$ sudo chmod 755 gphoto2-updater.sh
$ sudo ./gphoto2-updater.sh
$ sudo apt-get install cups libcups2-dev
$ sudo apt-get install python3-opencv

$ pip3 install pygame-menu==2.3.3

$ git clone https://github.com/werdeil/pibooth.git
$ cd pibooth

""" With the option you want as describe in the installation instructions"""
$ sudo pip3 install -e .[dslr,printer]

"""Now to launch"""
- Close the command promp windows or write cd+enter
- Write pibooth+enter and it will start.

anxuae, 2020-05-19:
Thanks for helping ;-)

The line `pip3 install pygame-menu==2.3.3` is now useless since the integration of `pygame-menu` v3 is done.

TheSuperpapi, 2020-05-19:
Ok, instruction updated :)

anxuae, 2020-05-19:
@Antoine125ys can you close this issue if the problem is solved.

thanks

closed 2020-05-25 by TheSuperpapi