pibooth/pibooth #145: Touchscreen not work on start with full screen mode

sravel, 2020-05-14:
**Describe the bug**
When pibooth run directly on fullscreen, the touchscreen doesn't work, bug if you open menu, then exit menu it's good and work again

**Steps to reproduce the bug**
1. start pibooth on full screen with touchscreen.
2. try to take photo
3. Open menu with button then exit
4. touchscreen work !!!

**Environment**
 - OS  Raspbian Buster
 - Camera. Rpi camera + Nikon D90
 - Pibooth version last master

anxuae, 2020-05-15:
Hi @sravel ,

I will try to reproduce it soon, with my 7'' touch screen.
I will keep you informed.

TheSuperpapi, 2020-05-19:
Hello,

I just tryed after update (developing version) and have the same result when start fullscreen, need to press esc with keyboard to enter setup and esc again to exit setup and it works.

closed 2020-05-22 by anxuae

TheSuperpapi, 2020-05-22:
Big thx 👍😉

commit 7e056070 (2020-05-22): Fix #145: invisible cursor (don't use pygame.mouse.set_visible)