pibooth/pibooth #147: LED flash at the same time of the photo

QuickPhilWoo, 2020-05-21:
Hello, I have a PI4-4Gb. I am new to the field. 
Thanks for this script which is really great! Everything works perfectly but I would like, if possible, to have 2 LED strips that light up at the time of the photo (like a flash) in order to have a well-lit photo.
I already have a relay module which is ready which I can supply via the GPIO ports of the Raspberry but I do not know how to order via a script my LED strips at the same time of the photo. Would you be so kind as to "enlighten me" please (lol). It would be necessary to modify the script and tell me on which port of the GPIO I should connect my relay module. 
Thanks in advance to the contributors. 
Yours.

anxuae, 2020-05-22:
Hi @QuickPhilWoo 

Have a look to the plugin [ibooth-extra-lights](https://github.com/pibooth/pibooth-extra-lights), it may fit to your need.

Note that currently plugins are only available for the developing version of `pibooth`. Have a look to how [install a developing version](https://github.com/pibooth/pibooth/blob/master/docs/dev.rst)

QuickPhilWoo, 2020-05-23:
Hello and thank you for your answer.
Indeed, I had not seen your intervention on the site.
This is exactly what I need. Thank you very much ;-)
Yours,

closed 2020-05-23 by anxuae

QuickPhilWoo, 2020-05-26:
Hello,

sorry to disturb you but when I install "upgrade", I get an error message:

  "Could not find a version that satisfies the requirement pibooth> = 2.0.0 (from pibooth-extra-lights) (from versions: 0.0.3, 0.0.4, 0.0.5, 0.0.6, 0.0.7, 0.0. 8, 0.0.9, 1.0.0, 1.0.1, 1.0.2, 1.0.3, 1.0.4, 1.1.0, 1.1.1, 1.1.2, 1.1.3, 1.1.4, 1.2.0)
No matching distribution found for pibooth> = 2.0.0 (from pibooth-extra-lights) "

Apparently, it does not find the right version on my Raspberry in PIBOOTH mode.

Can you help me ?

Yours,

Philippe DROOGMANS

Closed #147<https://github.com/pibooth/pibooth/issues/147>.

werdeil, 2020-05-26:
You'll need to install the development version of pibooth or wait for the version 2.0.0 to be release, I don't know when we will release it though.

QuickPhilWoo, 2020-05-26:
Ok Tank’s 😉

Best regards

Philippe DROOGMANS

[logo signature Outlook]

Tel : 0475/87.77.67
www.quicksoft.be