pibooth/pibooth #149: pibooth application locking USB devices

thibault-78, 2020-05-25:
Hello
Thx for the great work you have made
I'm trying to make working CV for preview and Gphoto2 for capturing)
-> based on what you already have implemented

issue seems related to USB lock:
When pibooth is  launched in CV mode : other USB device can not accessed anymore

To reproduce the bug:
-modif of init.py (see bellow)
-- new hybrid mode is the same than CV
-Result
-- not able anymore to take picture with gphoto2 in parallel of pibooth app 

> gphoto2 --capture-image
>                                                                                
> *** Erreur ***
> An error occurred in the io-library ('Impossible de communiquer avec le périphérique USB'): Impossible d'utiliser l'interface 0 (Périphérique ou ressource occupé). Assurez-vous qu'aucun autre programme (gvfs-gphoto2-volume-monitor) ou module du noyau (comme sdc2xx, stv680, spca50x) n'utilisent le périphérique et que vous y avez accès en lecture et écriture.
> Erreur : Acquisition d'image impossible.
> Erreur : Acquisition impossible.
> *** Erreur (-53 : « Impossible de communiquer avec le périphérique USB ») **

Not sure that the issue is coming form pibooth apps or OpenCv apps

Code I woud like to make it work
init.py
```
elif gp_camera_connected() and cv_camera_connected():
        LOGGER.info("Configuring hybrid camera (OpenCV + gPhoto2) ...")
        #cam_class = HybridCamera2
        cam_class = CvCamera
        pkill('*gphoto2*')

```

hybrid.py
```
from pibooth.controls.camera.opencv import CvCamera
class HybridCamera2(CvCamera):

    """Camera management using a Cv camera for the preview (better
    video rendering) and a gPhoto2 compatible camera for the capture (higher
    resolution)
    """

    IMAGE_EFFECTS = GpCamera.IMAGE_EFFECTS

    def __init__(self, *args, **kwargs):
        CvCamera.__init__(self, *args, **kwargs)
        self._gp_cam = GpCamera(*args, **kwargs)
        self._gp_cam._captures = self._captures  # Same dict for both cameras

    def _post_process_capture(self, capture_path):
        """Rework and return a Image object from file.
        """
        return self._gp_cam._post_process_capture(capture_path)

    def capture(self, filename, effect=None):
        """Capture a picture in a file.
        """
        self._gp_cam.capture(filename, effect)

        self._hide_overlay()  # If stop_preview() has not been called

    def quit(self):
        """Close the camera driver, it's definitive.
        """
        super(HybridCamera2, self).quit()
        self._gp_cam.quit()
```

anxuae, 2020-05-25:
Hi @thibault-78 

Can you copy-paste the full python traceback?
Also give me the web camera and DSLR name/brand that you are using?

Note that both classes `GpCamera` and `CvCamera` are trying to initialize the first found camera. The possible reason that I'm thinking is: if your DSLR/Webcam is compatible with both libraries (OpenCV and libgphoto2), maybe the both classes are trying to connect to the same device?

thibault-78, 2020-05-25:
Hello @anxuae ,
- Logitech, Inc. Webcam C930e
- Canon, Inc. EOS 60D

You are right,
Since my post I have made other tests that confirm your idea
Gphoto2 working only with my DSLR Canon 
OpenCV with both
Based on that fact, I just changed the init order in the __init__ hybrid class in my code and make it work :)
-> Probably a little bit dirty because if both class try to initialise the first they found it will not be predictive (with other products ) but as gphoto2 is usually used for better camera (and more restrictive) it should work on most of the case
Your code seems  

```
def __init__(self, *args, **kwargs):
        self._gp_cam = GpCamera(*args, **kwargs)
        CvCamera.__init__(self, *args, **kwargs)
        self._gp_cam._captures = self._captures  # Same dict for both cameras
```
Will you push your code in the next release ? I will be happy to test
-> For now I only installed the 1.2.0 official release

thibault-78, 2020-05-28:
Hello @anxuae 
Tryed with last dev version and your code:
Still the same issue : 
    self._cam.init()
gphoto2.GPhoto2Error: [-53] Could not claim the USB device
The workaround of init gpho2 before cv is still working

```
def __init__(self, *args, **kwargs):
        self._gp_cam = GpCamera(*args, **kwargs)
        super(HybridCvCamera, self).__init__(*args, **kwargs)
        self._gp_cam._captures = self._captures  # Same dict for both cameras
```

Regards

werdeil, 2020-05-30:
My only concern is that we can't test it. I'm not very fan of adding code that we can't test.

@anxuae I have a logitech webcam but we need the DSLR to make some tests.

anxuae, 2020-05-31:
I have tested the code with the webcam of may mac and a Nikon DSLR camera.

@thibault-78 did you try with the developing version on a raspberry-pi (Logitec Webcam + Canon DSLR)?

thibault-78, 2020-05-31:
Hello @anxuae 
yes  I tested your code with a complete reinstall:
- R-Pi 4B
- Raspbian
- last dev version + your changes
- Logitech, Inc. Webcam C930e
- Canon, Inc. EOS 60D
 I had to manually install Python 3.7.7 and last pigame 1.9.6 to make it work : Could be interesting to mention it on the install part section when you will release the V2 version. 
- if you need someone to test when it will be release I can help

As I mentionned, I had to change the init order of the camera on your code to make it work for me

```
def __init__(self, *args, **kwargs):
        self._gp_cam = GpCamera(*args, **kwargs)
        super(HybridCvCamera, self).__init__(*args, **kwargs)
        self._gp_cam._captures = self._captures  # Same dict for both cameras
```

Other thing that can help beginers like me: I though that the Id mentioned for LED / Push buton were GPIO number and  it is PIN number. Maybee in the config file it can help to add in the comment that it is not GPIO Number

Next steps for me with your app: 
- power on / power off button (as mentioned in other threads) as I don't want to use a keyboard (and make the thing safer)
- I also would like to add another sceen (possible with RPI4) and use it in // of pibooth to have a slide show of the pictures taken during the event (people waiting could have a distraction)
- Build the box

werdeil, 2020-05-31:
Ok let me a month or two to buy a second setup and make some tests.

Out of subject but about the 

> power on / power off button (as mentioned in other threads) as I don't want to use a keyboard (and make the thing safer)

Do you know that you have an option to launch pibooth at startup of the raspberry? We use it on our first setup and we don't have any keyboard.

thibault-78, 2020-05-31:
Hello @werdeil 
Yes I know I use it :). 
2 objectives for the power on/off button: 
- clean power off.
- In Hybrid mode I experienced with my tests some problems: if my canon went out of battery or not reacheable (sleep or could be other things) you have a "oups something went wrong" : In that case I want to make it work only with the webcam: The button will stop/start the system and pibooth will work again in OPenCV mode :)

anxuae, 2020-11-14:
Hi @thibault-78 ,

The PR is now merged, the feature will be available in pibooth 2.0.2

Thanks for reporting.
anxuae

closed 2020-11-14 by anxuae

Refurb, 2024-01-22:
Hi, I still have this issue. Running Pibooth 2.0.7. What can I do?