pibooth/pibooth #150: Printer N° count

TheSuperpapi, 2020-05-28:
Hello there,
As usual thanks for this amazing work and you overall support :)

One idea could be a printer count:
Implement somewhere a file with the printer count of the session (maybe when someone is pushing on print option)
It could help to know where we are with the "printer consumables"

### just a question / not the same subject ###
Today the final picture format is 8/12
Could you help me to find how can I change it for 4/6?

Have a nice day,
Best Regards.

werdeil, 2020-05-28:
Hi @TheSuperpapi, we have something implemented for user of selphy printer (which have a limited number of prints) through the `'PRINTER', 'max_pages'` config setting. This avoid sending print event to an empty printer.

But what you are asking is more a logging to get the infos afterward. The more I think about it the more I find it an interesting thing to keep. @anxuae what do you think? It can't be interesting for a pibooth owner to know how many prints were asked for a session.

About your side question I don't understand your issue, 8x12 and 4x6 are the same ratio, the final picture is 4x6 at a 600dpi so 2400x3600 pixels. Is it that your printer resolution is 300dpi?

TheSuperpapi, 2020-05-28:
Hi,

Counter:
In fact i'm setting up a dnp DS 620 with a capacity of 400 print.
But there is no counter on the printer and I can't know of 100 where used or 350 accept with estimation of the paper roll thickness visually.

For the second little question, i's more an issue for me.
Ratio is the same as you said but when I print something is missing on the final picture compare to the view, i have a kind of crop... 

If I correct the last picture to 4x6 instead of 8x12, I don't have this issue, that is a bit strange for me, I don't understand how it is possible!!! I'm looking the soft side and printer option side but, don't find this issue.

If it not clear I'll open a specific point to ask for help or idea on that with a video.

anxuae, 2020-05-29:
So, as I understand is to keep the number of printed picture somewhere to make it persistent after reset.
Maybe an elegant way to read it and rest it could implemented in the menu. Currently the function is a little bit hidden as the counter reset is done when the menu is opened.

anxuae, 2020-05-29:
I can make a proposal if you want

werdeil, 2020-05-30:
> an elegant way to read it and reset could implemented in the menu

Yes but this is not what I meant: I meant something that can be accesses outside pibooth and that cannot be reseted by pibooth itself. But this may not be what @TheSuperpapi wants.

TheSuperpapi, 2020-05-30:
In fact thé final goal is to know, after à selphy booth évent  if i used 300 paper or 50 papers.
If the coûnter is à total without reset or or total of thé selphy session, it's quitte thé same because it allow to calculate. 😁

werdeil, 2020-06-10:
Ok then @anxuae you can make your proposal to have the printer count and resets through the menu.

Ideally I would add more stats to this submenu: number of photos taken, number printed, number forgotten... But it is up to you ;-)

anxuae, 2020-06-22:
This feature is included in the v2, now released.

closed 2020-06-22 by anxuae

TheSuperpapi, 2020-06-23:
Big Thanks to you guys.
Not at home for still 1.5 month and can not test.
Will do that when I Will be at home 😉