pibooth/pibooth #151: Minimum timer on the last screen "thank you"

TheSuperpapi, 2020-05-28:
Hi again.

Will be more diffcult for me to explain with my english level but:   :)

Maybe it could help to gave a more linear result if you add a kind off minimum time on the last screen

For exemple, on a pi 4B 4GO, the last screen one where it's written "thanks you" just appear 0.5 s and the screen before, the "processing" one take at least 5s.

Maybe it could be more linear to keep this "thank you" screen 3 or 4 sec minimum to give time to see it and be more linear with the time of processing just before.

:-)

werdeil, 2020-05-28:
Hi @TheSuperpapi, indeed the thank you screen is actually meant to last 0.5s (there is a timer for that):
`self.finish_timer = PoolingTimer(0.5)`

Maybe we can increase it but not to 3 or 4s, it is very long for a "useless" screen, maybe 1s?

For me this screen is only here to avoid any misclick between the print screen and the start screen. On the contrary processing is only limited by the computer process.

If you really want to have a customizable timer I think this can be done through a plugin with the new plugin process.

@anxuae What do you think?

TheSuperpapi, 2020-05-28:
Hummm,

I need to try changing the 0.5 to 1 or 2s max with a "funny thank you screen" or to keep it but remove the text and image (just have a black screen)

I'll make some test :)

Thanks again for your greet support. :-)

closed 2020-05-28 by TheSuperpapi