pibooth/pibooth #152: Need printing help or suggest

TheSuperpapi, 2020-05-29:
Hello there,

As I mentioned in a previous post, I open this point for help or suggestions related to printing issues.

See the link below:
On the left side, is a print via gimp without specific fit options.
On the right is the result via pibooth.

https://ibb.co/tp0PNNV

Printer is DNP ds620 printing on 4x6 via cups 300x600 dpi or 300x300dpi (same result)

Is someone have a idea for me? :-)

anxuae, 2020-05-29:
It seem that the printer is not well configured in CUPS server, it should be borderless.
Have you try to change the default parameters of the printer at http://localhost:631/printers ?

TheSuperpapi, 2020-05-30:
Hello,

Yes, I have tried thé 3 proposed solution. 
Shrink / crop / extend. 

I have tried with my laser printer and it's ok. 

So I need to check if i can read and update thé ppd file of this printer or i will have to build specific template to avoid this crop to be visible. 

Another solution is to update i suppose both sizing and factory to fit my printer issue but it's à bit hard for me 😁

I'll close this topic soon, just would like to see if another found à solution with dnp ds620 because it's the mainly used printer for selphybooth

Thanks Again for your support. 👍

anxuae, 2020-05-31:
Hi @TheSuperpapi , no problem this is the place where question can be put.

Did you try to install [Gutenprint](http://gimp-print.sourceforge.net/p_Download.php)? It may add some extra parameters to configure the printer.

See https://raspberrypi.stackexchange.com/questions/40722/how-to-install-cups-driver-gutenprint

> Another solution is to update i suppose both sizing and factory to fit my printer issue but it's à bit hard for me 😁

The issue seems external to the `pibooth` application as there is no margin added by the software around the background image. Maybe you can try to set `[PICTURE]margin_thick=0` in case of a bug is hidden somewhere in this option.

TheSuperpapi, 2020-06-02:
I close the issue, had no time to work on but Will try your suggestion 😁

Thanks à lot for your time.

closed 2020-06-02 by TheSuperpapi