pibooth/pibooth #153: Invalid resolution requested

moloco95, 2020-05-31:
bonjour,
J'ai tout suivi  mais je suis tombé sur ses erreurs
Y a t'il un moyen d'arranger sa? 
Merci 
Hello,
I followed everything but I came across his mistakes
Is there a way to fix it?
Thank you

pygame 1.9.4.post1
Hello from the pygame community. https://www.pygame.org/contribute.html
[ INFO    ] pibooth           : Starting the photo booth application...
[ INFO    ] pibooth           : Configuring Picamera camera ...
Traceback (most recent call last):
  File "/usr/local/bin/pibooth", line 10, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.7/dist-packages/pibooth/booth.py", line 741, in main
    app = PiApplication(config)
  File "/usr/local/lib/python3.7/dist-packages/pibooth/booth.py", line 447, in __init__
    config.getboolean('CAMERA', 'delete_internal_memory'))
  File "/usr/local/lib/python3.7/dist-packages/pibooth/controls/camera/__init__.py", line 39, in get_camera
    return cam_class(iso, resolution, rotation, flip, delete_internal_memory)
  File "/usr/local/lib/python3.7/dist-packages/pibooth/controls/camera/rpi.py", line 53, in __init__
    self._cam.resolution = resolution
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 2271, in _set_resolution
    "Invalid resolution requested: %r" % (value,))
picamera.exc.PiCameraValueError: Invalid resolution requested: PiResolution(width=1934, height=2464)

moloco95, 2020-05-31:
I have camera v1

werdeil, 2020-06-01:
Bonjour, 

Ce problème apparait quand on met une resolution non compatible avec la picamera, il faut mettre une résolution supportée par la camera, la liste est sur la page suivante: https://picamera.readthedocs.io/en/release-1.12/fov.html

Essayez de changer le ratio de la preview (1934, 2464) qui ne semble pas compatible avec la camera v1 par un du tableau dans le lien.

moloco95, 2020-06-01:
Merci, ducoup, je vais sur la programmation, de la caméra et je change sur la ligne ? 
En tout cas Merci beaucoup

werdeil, 2020-06-01:
Il suffit de changer le réglage dans la configuration du `pibooth`, soit en lançant la commande 
> pibooth --config

soit en modifiant directement le fichier `~/.config/pibooth/pibooth.cfg`

moloco95, 2020-06-01:
j'ai essayé cela ne fonctionne pas 

c'est bien ici qu'il faut changer dans parsey.py   ?
("CAMERA",
        odict((
            ("iso",
                (100,
                 "Adjust for lighting issues, normal is 100 or 200 and dark is 800 max",
                 None, None)),
            ("flip",
                (False,
                 "Flip horizontally the capture",
                 None, None)),
            ("rotation",
                (0,
                 "Rotation of the camera: 0, 90, 180 or 270",
                 None, None)),
            ("resolution",
                ((1934, 2464),
                 "Resolution for camera captures (preview will have same aspect ratio)",
                 None, None)),
            ("delete_internal_memory",
                (False,
                 "Delete captures from camera internal memory (when applicable)",
                 None, None)),
        ))

j'ai changé le 1934, 2464 rien n'y fait jai toujours 

    config.getboolean('CAMERA', 'delete_internal_memory'))
  File "/usr/local/lib/python3.7/dist-packages/pibooth/controls/camera/__init__.py", line 39, in get_camera
    return cam_class(iso, resolution, rotation, flip, delete_internal_memory)
  File "/usr/local/lib/python3.7/dist-packages/pibooth/controls/camera/rpi.py", line 53, in __init__
    self._cam.resolution = resolution
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 2271, in _set_resolution
    "Invalid resolution requested: %r" % (value,))
picamera.exc.PiCameraValueError: Invalid resolution requested: PiResolution(width=1934, height=2464)

en tout cas merci beaucoup pour votre aide

anxuae, 2020-06-08:
Is this issue fixed?

moloco95, 2020-06-08:
Bonjour, non ducoup j'ai la caméra 2.1 qui arrive

anxuae, 2020-08-11:
Hi @moloco95 , do you still have issue about resolution?

anxuae, 2020-08-18:
Please, feel free to re-open the issue if you're still facing this issue.

closed 2020-08-18 by anxuae