pibooth/pibooth #154: Pibooth v2.0.0 / final_image_delay not working

thibault-78, 2020-06-01:
with last dev version V2.0.0 , it is not possible to hide the last picture.
The value of final_image_delay seems not been taking in account

thibault-78, 2020-06-05:
After test with a printer:
-> this settings seems to be used on the forget / print page and not to the main page anymore
Would it be possible to implement the same mecanism on the main page?

werdeil, 2020-06-05:
Indeed found the bug, I just pushed a fix, but I think @anxuae will find a better one, as mine is refreshing the screen everytime.

thibault-78, 2020-06-05:
Hello @werdeil 
Many thanks for your reactivity 
It is working almost fine :)
-> When I came in the main page the LED "print" is still blinking

I also notice a stange thing : i think it is linked to that feature and/or V2
in V1.2 : the main page consuming almost 0% CPU (at start or when displaying last picture),  when the picture is hidden it sudenly goes to 25% (i'm on a RPI4b)
In V2 : the main page alway took 25%

werdeil, 2020-06-05:
Ok I'll look at the led, I think it is the same kind of issue.

About the CPU usage: can you test with my latest commit? It shall be the same as 1.2.0.

The 25% issue is that we refresh the screen continuously after the timeout. We have this issue since a long time but I'll try to find a better solution.

thibault-78, 2020-06-05:
In V2 with your last commit (without printer)
- at launch 25% 
- took picture 
- main page + picture 4%
- timeout
- main page without picture 25%

= same behavior for both versions
-> except at the boot: v2 25% compare to the 4% on V1.2 (if I well remember)

thibault-78, 2020-06-06:
Hello @werdeil 

> The 25% issue is that we refresh the screen continuously after the timeout. We have this issue since a long time but I'll try to find a better solution.

In fact it took all the CPU of a core (seems to share the same core than  xorg and both process took 100% of the core when we are on the main page)

As suggested, problem seems to be in the loop:
If have a very dirty solution : sleep 1 sec 
-> it could  add a lag of max 1sec when a button is pushed  but CPU decrease very significantly 100% of the core(pibooth+xorg process)  to  less than 8% (pibooth+xorg process)  

in booth.py

```
import time 
            while True:
                events = list(pygame.event.get())

                if self.find_quit_event(events):
                    break

                if self.find_fullscreen_event(events):
                    self._window.toggle_fullscreen()

                event = self.find_resize_event(events)
                if event:
                    self._window.resize(event.size)

                if not self._menu and self.find_settings_event(events):
                    self.leds.off()
                    self._menu = PiConfigMenu(self._window, self._config)
                    self._menu.show()
                    self.leds.blink(on_time=0.1, off_time=1)
                elif self._menu and self._menu.is_shown():
                    self._menu.process(events)
                elif self._menu and not self._menu.is_shown():
                    self.leds.off()
                    self._initialize()
                    self._machine.set_state('wait')
                    self._menu = None
                else:
                    self._machine.process(events)
                    # UGGLY delay !! 
                    time.sleep(1)

```

anxuae, 2020-06-06:
Hi @thibault-78 

Can you retry with the fix that I've just pushed?

thibault-78, 2020-06-06:
Hello @anxuae,
Everything seems working : 
-> print button stop blinking when picture is hidden
Many thanks

commit d8155c7d (2020-06-06): Fix #154: final_image_delay != 0 and printer LED
closed 2020-06-06 by thibault-78