pibooth/pibooth #155: Ne démarre pas après avoir installé la version en développement...

QuickPhilWoo, 2020-06-02:
Raspberry PI4 - 4 GB - Rasbian - Pas d'appareil Photo - camera RPI

Bonjour, 

après avoir installé la version en développement, quand je veux démarrer l'application (pibooth dans la console), la console m'affiche un message d'erreur (voir ci-dessous) ...
N'est-ce pas dû à ma version de Python ? Comment détourner le problème ? 
D'avance merci pour vos réponses ;-)
Bien à vous,

QuickPhilWoo, 2020-06-02:
Voilà le code d'erreur 👍 
pygame 1.9.4.post1
Hello from the pygame community. https://www.pygame.org/contribute.html
Traceback (most recent call last):
  File "/usr/local/bin/pibooth", line 11, in <module>
    load_entry_point('pibooth', 'console_scripts', 'pibooth')()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 489, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2793, in load_entry_point
    return ep.load()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2411, in load
    return self.resolve()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2417, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/home/pi/pibooth/pibooth/booth.py", line 26, in <module>
    from pibooth.config import PiConfigParser, PiConfigMenu
  File "/home/pi/pibooth/pibooth/config/__init__.py", line 4, in <module>
    from pibooth.config.menu import PiConfigMenu
  File "/home/pi/pibooth/pibooth/config/menu.py", line 7, in <module>
    import pygame_menu as pgm
  File "/usr/local/lib/python3.7/dist-packages/pygame_menu/__init__.py", line 66, in <module>
    import pygame_menu.themes
  File "/usr/local/lib/python3.7/dist-packages/pygame_menu/themes.py", line 386, in <module>
    THEME_DEFAULT = Theme()
  File "/usr/local/lib/python3.7/dist-packages/pygame_menu/themes.py", line 198, in __init__
    bool, pygame.vernum.major == 2)  # type: bool
AttributeError: 'tuple' object has no attribute 'major'

thibault-78, 2020-06-05:
Hello @QuickPhilWoo 
Pour avoir installé la version de dev il y a quelques jours il te faut installer manuellement la version python 3.7.7 et la version 1.9.6 de pygame.

anxuae, 2020-06-06:
The bug is coming from `pygame-menu` <= v3.0.3 , it will be fixed in the next release

QuickPhilWoo, 2020-06-08:
Bonjour,

Merci beaucoup pour votre réponse.

Je vais voir ça …

Bien à vous.

Philippe DROOGMANS

asbjornstokka, 2020-06-12:
After following the advice I still got an: "ModuleNotFoundError: No module named 'pygameMenu'" Error. 

Fixed it by changing the import from "pygameMenu" to "pygame_menu".

anxuae, 2020-06-22:
The version 2.0 is now released.
(re-open it if the bug is still present)

closed 2020-06-22 by anxuae

grenagit, 2020-06-26:
On a fresh installation of **Pi OS** (previously called Raspbian) on **Raspberry Pi 3B+**, I have the same bug when installing!

I forced the installation of the latest version of pygame with `sudo pip3 install pygame==1.9.6` to fix the problem.
The problem probably comes from `setup.py` where `Pygame>=1.9.4` should be `Pygame>=1.9.6` ?

QuickPhilWoo, 2020-06-29:
Hello,

Do you see : https://github.com/pibooth/pibooth/issues/142 ?

Best regards

QuickPhilWoo, 2020-06-29:
Hello and thank you ! with your command line it works!

anxuae, 2020-06-29:
Hi,

I made a fix that will be available in the next `pibooth` release.

It seems that `pygame` is not correctly upgraded because 2 different versions of `pygame` are defined in both `pibooth` and `pygame-menu`

QuickPhilWoo, 2020-06-30:
Hi,

Ok it’s perfect ! Thanks

Best regards

QuickPhilWoo, 2020-06-30:
Hello,
I have a Pi 3B, Pibooth works but I can't find the .config ...
I have a friend who uses it too and he finds the .config in: / home / pi.
I don't have it !
No matter how much I search, it can't find ...
Could you tell me where to find the Pibooth configuration file please?

[Une image contenant capture d’écran  Description générée automatiquement]

Philippe DROOGMANS

QuickPhilWoo, 2020-06-30:
Hi,

It’s ok ! The folder is hide !

Sorry !

Best regards

Philippe DROOGMANS