pibooth/pibooth #160: aide pour plugin

sravel, 2020-06-18:
Bonjour,
Pour le développement d'un plugin, je voudrais trouver une solution pour l'activer/désactiver via le menu.
J'ai donc ajouté un params
activate = True or False
Comment puis-je faire pour "capturer" quand le fichier de config est mis a jours, pour voir si la valeur est modifier via le menu ?

anxuae, 2020-06-18:
Hello @sravel ,

> I'm switching in english to permit non-french user reaction, if any.

Just call `cfg.get..` in your hooks is enough: the `cfg` object is automatically updated when the menu is closed.

werdeil, 2020-06-18:
And I would add that you cannot fully disable the plugin (the plugin will always be loaded and you will always enter all the hooks but you can have a setting that disable plugins actions in each hook.

sravel, 2020-06-19:
Ok thank,
I use the cfg that I give to my class, but I have to update it in each hook.
this is how i did:
https://github.com/sravel/pibooth-pimoroni11x7
first I get the cfg.get in the hook, then pass it to my class by the function, but it was redundant tu check if activate...
@werdeil yes I understood that the hooks are called, so it is my class which manages the display or not

anxuae, 2020-06-19:
Hello @sravel 

Some notes:

- Don't need to pass the `cfg` object in your class in each hook. This variable is a pointer, if you give it in the constructor of your class, you can access to its updated values at any time.
- To avoid checking `hasattr(app, 'pimoroni_11x7')` , you can create the object in `pibooth_startup ` even if the plugin is not activated
- Check if `activate=True` in the hooks, it avoid to keep an intermediate variable `self.activate`  in your class that you have to keep synchronized

I would have done your hooks this way:

```python

@pibooth.hookimpl
def pibooth_startup(app, cfg):
    """Create the PiboothPimoroni11x7 instance."""
    app.pimoroni_11x7 = PiboothPimoroni11x7(config=cfg)

@pibooth.hookimpl
def state_wait_enter(app, cfg):
    if cfg.getboolean('PIMORONI11x7',  'activate'):
        app.pimoroni_11x7.clear_and_write(message=cfg.get('PIMORONI11x7', 'wait_message'))

@pibooth.hookimpl
def state_wait_do(app, cfg):
    """
    Display wait view message.
    """
    if cfg.getboolean('PIMORONI11x7',  'activate'):
        app.pimoroni_11x7.wait_scroll(interval=0.05)

@pibooth.hookimpl
def state_choose_enter(app, cfg):
    if cfg.getboolean('PIMORONI11x7',  'activate'):
        app.pimoroni_11x7.clear_and_write(message=get_translated_text('choose'))

@pibooth.hookimpl
def state_choose_do(app, cfg):
    if cfg.getboolean('PIMORONI11x7', 'activate'):
        app.pimoroni_11x7.wait_scroll(interval=0.05)

@pibooth.hookimpl
def state_chosen_enter(app, cfg):
    if cfg.getboolean('PIMORONI11x7', 'activate'):
        app.pimoroni_11x7.clear_and_write(message="GO")
        app.pimoroni_11x7.wait_scroll(interval=0.05)

@pibooth.hookimpl
def state_preview_enter(app, cfg):
    if cfg.getboolean('PIMORONI11x7', 'activate'):
        app.pimoroni_11x7.preview_countdown()

@pibooth.hookimpl
def state_capture_enter(app, cfg):
    """Ready to take a capture."""
    if cfg.getboolean('PIMORONI11x7', 'activate'):
        app.pimoroni_11x7.flash()

@pibooth.hookimpl
def state_capture_exit(app, cfg):
    """A capture has been taken."""
    if cfg.getboolean('PIMORONI11x7', 'activate'):
        app.pimoroni_11x7.clear_and_write(message=get_translated_text('smile'))

@pibooth.hookimpl
def state_processing_enter(app, cfg):
    if cfg.getboolean('PIMORONI11x7', 'activate'):
        app.pimoroni_11x7.draw(smile_list)

@pibooth.hookimpl
def pibooth_cleanup(app):
    if cfg.getboolean('PIMORONI11x7', 'activate') and not app.pimoroni_11x7._DISABLE:
        app.pimoroni_11x7.clear()
        app.pimoroni_11x7.show()
```

sravel, 2020-06-22:
thanks for help, i make the modifications and all work !!!!

closed 2020-06-22 by sravel