pibooth/pibooth #163: Not able to launch after fresh install

Tabata975, 2020-06-28:
Hello
I'lm facing an issue when installing PiBooth

I donwloaded the PI desktop image and then follow all steps on the project page.
But when i want to lancu Pibooth i get the error
'AttributeError: "tuple' object has no attribute "major'

Before i see "Bool, Pygame.vernum.major == 2) #Type: Bool"

During the install i saw an error message "Pygame-menu 3.1.2 has requirement pygame>=1.9.6 but you will have Pygame Pygame 1.9.4 which is incompatible. 

I see Pygame is from the Python 2.7 folder where i would expect it from the newest 3.7. Could it be linked ?

Sorry for my few knowledge regarding what it can be.

Tabata975, 2020-06-28:
Hello
Finally found how to fix it
sudo pip3 install pygame --upgrade

if this can help

closed 2020-06-28 by Tabata975

Exitras, 2020-08-18:
Could you include this (sudo pip3 install pygame --upgrade) to the main install introduction? I was facing the same issue and it took me quite some time to solve it.

anxuae, 2020-08-18:
Hello @Exitras 

Thanks for your feedback. This change will be done in the next release (the pygame version is fixed in the `setup.py` )