pibooth/pibooth #164: Resolution constraint on camera photos

supermikeedwards, 2020-08-17:
Hi,
First of all, thanks a lot for Pibooth, it's incredible work ! I have been searching topics related to this one and couldn't find an answer so posting a new one. Sorry if some of my assumptions aren't correct, I'm not at all an expert in Python. 

I am running the hybrid mode and I can see in the logging that the program is recognizing that. I was however surprised to see the size of the raw photos after taking a few and found them to be very small in size. I initially thought there was an issue and that the DSLR camera wasn't recognized and therefore the photos in the raw folder where those coming from the Picamera. However, after digging a little deeper, I think it's coming from the resolution setup. My Canon 1300D is taking photos at 5184*3456 pixels and they are then being resized down to 1920*1080 pixels as set in the config file. I tried increasing the value in the config file but the program fails which I think is probably linked to the fact that I'm going over the max resolution of the Picamera V2 that I have installed. 

1920*1080 looks fine on a 15*10cm print, but I would like the raw files to be the full resolution ones so that I can share the high quality files. Is there no way to do that ? 

Thanks
Mike

werdeil, 2020-08-18:
Hi @supermikeedwards,

On DSLR with gPhoto we made the choice to let the choice of the raw photo resolution on the DSLR side so you shall be able to increase the image quality of the raw images on the camera side. 

However as the raw images are getting bigger the processing time will also increase. In this processing time we have the downloading of the raw photos, the building of the final picture with a better resolution. You'll need to find the better compromise between your acceptable processing time and the wanted raw resolution.

Tell us if you manage to change the setting on the 1300D side, we use a 1100D on our version, it shall be very similar.

supermikeedwards, 2020-08-18:
Hi Vincent,
Thanks a lot for your answer. I checked on the DSLR and I'm already at maximum JPEG quality (5184x3456 resolution). But when I look at the photos in the raw folder they are all at 1920*1080 resolution which matches the config file. After looking through the code, for me this is coming from : 
_# Resize to fit the resolution
        image = image.resize(sizing.new_size_keep_aspect_ratio(image.size, self.resolution, 'outer'))_
Which is located in the Gphoto python file.
I changed that line to a comment and indeed the resolution increased to 5184x2916 as the photo has been cropped down from it's original height to match the screen size ratio. This is a lot better and I can see the difference in quality between before and after if I take the same photo with and without that line. 
However, when I compare the original file that I take from the DSLR internal memory card and compare it to the saved photo, the resolution is similar but the file saved in the raw folder is a lot smaller (500-600kb VS 5-6Mb on the DSLR memory card), so I still think the post processing of the image file is decreasing the quality. 
I'm quite happy with my workaround but I'm still wondering whether it's possible to have in the raw folder the actual file from the DSLR and not the cropped/resized one ?
Thanks
Mike

werdeil, 2020-08-20:
Indeed you are right, after looking back at the history we chose to store the downsized raw version in the raw folder as the untouched raw version is always available in the SD card of the camera.

I will have a look to see what we can do to this.

supermikeedwards, 2020-08-21:
Hi,
Thanks for your answer. Just in case anyone else is looking to do the same thing as me, be careful if you remove the resize line like I did. On a Raspberry Pi 3, the image reprocessing when you do a series of 4 photos overloads the processor (probably due to the photos being that much bigger) which leads to (a) a lot of lagging when you ask the user if they want to print (b) for some reason I can't explain your next series of photos will make the program crash. The photos are taken by the DSLR but aren't imported into the folder for display. 
Mike

anxuae, 2020-08-21:
Hi @supermikeedwards,

Can you give me the console logs when the program crash? I guess that there is a memory issue because of handling 4 images with very high resolution, thus it overloads both CPU and memory.

anxuae

supermikeedwards, 2020-08-21:
Hi Anxuae,
I would love to be able to give the console logs, but I don't know how to do that. I know how to look at python code and make adjustments, but I have a lot of trouble debugging correctly !
My assumption is based on the CPU and memory activity log where I can see that both spike up during the image processing.
THanks
Mike

anxuae, 2020-09-23:
Hello @supermikeedwards ,

Any news on this topic?

Thanks
Anxuae

anxuae, 2021-01-31:
Hi @supermikeedwards, I propose to close this issue as it seems related to your setup/camera. If you are still stuck on it, please feel free to re-open it and give us more feedback.

BR
anxuae

closed 2021-01-31 by anxuae