pibooth/pibooth #165: Where is the directory of installation ?

Astryder, 2020-08-19:
Hi folk !

I would like : 
-  to add some command between differents action of push button in python.
-  to autostart this programm at boot of my raspberry 

But I don't know where is the directory of this installation to make some modification :(

I don't have any folder about pibooth in /home/pi/

Can you help me ?

Thx for a alot

best regards

Astryder

anxuae, 2020-08-19:
Hello @Astryder 

`pibooth` is installed with all other python library.
You can easily find it by typing this command in a shell:

``
python -c "import pibooth;print(pibooth.__file__)"
``

You can enable an option in the config to start `pibooth` at raspberry-pi startup.

To add custom commands, the best is to develop your own plugin : https://github.com/pibooth/pibooth#customize-using-plugins

Astryder, 2020-08-20:
Thx a lot for your answer !

The right directory is 
/usr/local/lib/python3.7/dist-packages/pibooth

Thx forthe tips about autostart (I didn't see befoire :) )

I follow your advise about my own plugin

closed 2020-08-20 by Astryder