pibooth/pibooth #166: Long button press required

Exitras, 2020-08-20:
Hi, I upgraded to 2.0.0 and like it a lot. Many thanks for the great work!
I finally don't experience the error anymore that the menu accidentally opens when using one of the buttons. However, would it be possible to eliminate the option to open the menu by hitting both buttons at the same time? I don't see any use of it and the risk of someone entering the menu by accident is way to high (as they don't know how to exit and start changing random things such as enabling debug mode - it actually happened during a wedding last week).

Using 2.0.0 right now, I need to press the buttons for almost one second to get some inputs which lead to confused guests. Some friends visited me this week and they all complained about the unexpected long button press time. Is there a way to decrese this time (preferably in combination with the disabled function to enter the menu by pressing both buttons).

**Environment**
 - Raspberry Pi 4
 - OS [e.g. Raspbian Buster]
 - Camera [Canon EOS 2000D]
 - Pibooth version [2.0.0]

ttette, 2020-08-20:
Hi, you need to change the debounce delay value to something smaller:

ttette, 2020-08-20:
in the config file > section CONTROLS

[CONTROLS]
# How long to debounce the hardware buttons in seconds
debounce_delay = 0.3

i think i used 0.01, which seems to work better

Exitras, 2020-08-20:
Hi @ttette, that works well, thank you 👍 
Do you also know how to make sure no one can enter the menu by pressing both buttons simultaneously?

ttette, 2020-08-20:
No, sorry. I didn't even know it worked like that :D

anxuae, 2020-08-20:
HI,

The only way is to change the code in order to add an option in the configuration.

**Proposal:**

Add a `debounce_multi_delay` which permit to define the required hold time when several buttons are pressed.
Setting it to a very long time with avoid the menu to be opened using HW button1 + button2.

@werdeil, what's do you think about it?

anxuae

Exitras, 2020-08-20:
Hi @anxuae, this sounds like a good idea 👍 
But isn't there a quick fix as you proposed in #83 such as commenting this line: https://github.com/werdeil/pibooth/blob/e54a8f1dd494fadcacfaa33071aafda9ce2ecb73/pibooth/booth.py#L632
When I did that, I couldn't open the menu with the ESC button anymore which was a bit unfortunate. It would be ideal to just eliminate the option to open the menu with the pushbuttons :)

werdeil, 2020-08-20:
> It would be ideal to just eliminate the option to open the menu with the pushbuttons :)

I disagree as it is quite useful to have access to the setting menu without having to connect a keyboard to the pibooth.

> Add a debounce_multi_delay which permit to define the required hold time when several buttons are pressed.

@anxuae I agree with this except the name of the setting, I would use something more explicit like `multi_press_min_duration`.

Exitras, 2020-08-20:
Hi @werdeil, thanks for your comment :) 
Yes, you are right. It is rather helpful to have access to the setting menu without having to connect a keyboard. However, I brought my photo booth to a wedding last weekend and it happened quite some time that people accidentally entered the settings menu and didn't know how to close it again. It happened for example, when multiple people are in front of the booth and when choosing the layout, one person hits the "1 photo" button and another person the "4 photos" button simultaneously. I was a little stressed during this wedding as I had to check the booth regularly to make sure it is still working properly. 
It would be way easier for me to disable this option entirely during an event as guests should not be able to enter the menu anyway.

Exitras, 2020-08-21:
Hi again, sorry for spamming so much :) 
I think I found the solution that works perfectly fine for me. Commenting those two lines prevents entering the menu by pressing the two buttons. That's what I needed :) 
https://github.com/pibooth/pibooth/blob/0ca54f8bb284616ca875d90f743519b9caa85dd4/pibooth/booth.py#L241-L242

Cheers
Exitras

anxuae, 2020-09-23:
Version 2.0.1 released! It should be fixed now.
Please reopen if not.

closed 2020-09-23 by anxuae