pibooth/pibooth #167: How can I install correctly on Pibooth my Raspberry cam HD 12m px ?

Astryder, 2020-08-20:
I want to install this new Rpi Cam 

https://www.raspberrypi.org/products/raspberry-pi-high-quality-camera/

I buy it specially for this projet :)

But When I install it on Pi... all is blur...and I'm so disappointed :(

Then I try to change the initial resolution by the resolution of this camera is 4000 x 3000 but nothing better :(

What can I do ?

Thx a lot

Best Regard

moskate49, 2020-08-20:
Hello it could be an hardware issue? Did you test cam hd alone?? On my pibooth with the old cam module (v2) the enclosire of the camera pus on the front of she and change the focus . Did you check this? The focus of the cam is ok?

https://www.raspberrypi.org/products/raspberry-pi-high-quality-camera/

Astryder, 2020-08-20:
Thx for your answer

Actually it works with the initial resolution 1934, 2464

But I can have the real resolution about my camera

Do you know which resolution can I take with this camera what will accept by pibooth ?

Because 3000, 4000 or 3040, 4056 doesn't work ... :(

anxuae, 2021-01-31:
Some PR in picamera library to support HQ camera (not yet released):

- https://github.com/waveform80/picamera/pull/639
- https://github.com/waveform80/picamera/pull/638
- https://github.com/waveform80/picamera/pull/660

anxuae, 2021-06-06:
Hi @Astryder ,

After some tests, I can use the HQ Camera in full resolution (4056 x 3040 pixels), by increasing the GPU memory to 256Mb.

```bash
sudo raspi-config
```

Go to `Performance Options` > `GPU Memory` > `256`
Then reboot the Raspberry-Pi.

Can you try and give me your feedback?

anxuae, 2021-06-09:
The problem seems to be solved by increasing GPU memory allocation (see #234)
Feel free to re-open issue if you encounter problems

closed 2021-06-09 by anxuae