pibooth/pibooth #170: Need helps for text size changing

TheSuperpapi, 2020-08-23:
Hello Guys,

Could you help me to increase this text size?

[image]

Thanks for your help,
Papi

anxuae, 2020-08-23:
Hi @TheSuperpapi ,

Starting `pibooth` in debug mode (menu: General > Debug mode) let you see the reserved placeholders for texts.
All texts are automatically resized to fit the red rectangles.

You can try to split your text in more than 2 lines in order to have a better filling.

But, if the placeholder is not big enough for your needs, you have to change the code here below to increase the size of the red rectangle:

https://github.com/pibooth/pibooth/blob/b79ead1162e9ef546e1a7ef6178ce71a6286b454/pibooth/view/background.py#L270-L279

TheSuperpapi, 2020-08-24:
Hello Anxuae,

Thanks for your reply.
You are confirming to me that I was looking the right file / direction and that I understood the method of auto text resize vs rectangle.

I'm pretty happy, it means that I begin to understand a bit your hard work :)

I'm learning a lot by try to understand everything :)

So now, I have to understand this part of the code line by line, and update it.

I have 2 weeks before my wedding, think will be ok.

I'm preparing my success story in // that you can have a overview below :)

[image]

Thanks again,
Papi

closed 2020-08-24 by TheSuperpapi

werdeil, 2020-08-24:
Wow really nice setup ! Thanks a lot for sharing the photo, do you allow us to put it in the [wiki](https://github.com/pibooth/pibooth/wiki) page? If you have more photos to share we will be glad to add them too.

If you need help to change the line of code ask us, and maybe we can manage to have something that fits all needs in the code.

TheSuperpapi, 2020-08-24:
Hi Werdeil,
Yes for sure, you can share the pictures, and if you need more details about the setup (component list, etc) just ask :)
Here more pictures :-)

I'll try to update the code by myself, I like to try before to ask :), it's a big challenge because I have no knowledge be I'm able now to aderstand when I read, juste have to learn a lot about formulas, commands, etc :-)

[image]
[image]
[image]
[image]
[image]