pibooth/pibooth #171: Plugins not load when running pibooth as root

RAW4RMCS, 2020-08-24:
I'm trying to implement neopixels as a countdown for pibooth. The only thing is, i need to run the neopixels als root or else they don't work. But when i run `sudo pibooth` the plugin doesn't load.

**Describe the bug**
Plugins not load when running pibooth as root

**Steps to reproduce the bug**
create a simple plugin with hooks like ex; 
```
import pibooth
from pibooth.utils import LOGGER

__version__ = "1.0.0"

@pibooth.hookimpl
def state_wait_enter(app):
    LOGGER.info("Hello from '%s' plugin", __name__)
```

add plugin to `plugins` under `GENERAL` in `pibooth.cfg`
run `sudo pibooth`.
Nothing is getting logged in the console.

**Environment**
 - OS [Raspbian Buster]
- RPI [3b+]
 - Camera [Rpi camera]
 - Pibooth version [2.0.0]

anxuae, 2020-08-26:
Hi @RAW4RMCS ,

Running `pibooth` as root (sudo), will generate the configuration in the `home` directory of the `root` user.
That means the configuration shall be edited using `sudo pibooth --config` and the config files are located in `/root/.config/pibooth`.

You have probably defined the plugin in the configuration file used by the `pi` user. That's why your plugin is not loaded.

By the way, it is not safe to run a program with `root` rights. Maybe you can change some rights in your raspberry-pi to avoid that? For instance, if you use the SPI port to connect your neopixels (https://github.com/joosteto/ws2812-spi), just add the `pi` user to the `spi` group will avoid to run `pibooth` as root.

anxuae, 2020-08-27:
Does it solve your issue?

RAW4RMCS, 2020-08-27:
Yes thank you 😊

closed 2020-08-27 by RAW4RMCS