pibooth/pibooth #173: Animate last pictures during idle time, and gallery

HBNielsenDK, 2020-08-29:
Thanks for a very powerfull and stable software.

In the features section is listed "Animate last pictures during idle time". I can only get the last picture to show static, and only in a small size.
Is it possible to cycle the last 10 pictures in fullscreen (landscape)?
Also make access to a gallery so the guests can see all the lovely pictures?

Thanks

anxuae, 2020-08-29:
HI @HBNielsenDK ,

> In the features section is listed "Animate last pictures during idle time". I can only get the last picture to show static, and only in a small size.

Be sure that the option is enabled in the configuration (disabled by default).

> Is it possible to cycle the last 10 pictures in fullscreen (landscape)?

No, the animation concernes the captures of the last sequence (= photos taken and concatenated), that means you will see an animation only if you chose 2, 3 or 4 captures.

> Also make access to a gallery so the guests can see all the lovely pictures?

Some external programs can easily do the job, for instance: https://lychee.electerious.com (but I never try it)

anxuae, 2020-09-13:
Hello @HBNielsenDK ,

I seen that @clemenstyp as developed a lightweight gallery in PHP, I did not try it but it may fit to your needs. 

If enough generic and @clemenstyp is ok, we can think about an integrated web gallery for pibooth

anxuae, 2020-09-23:
I think that this topic can be closed now.
Please reopen it if you have remaining open points.

closed 2020-09-23 by anxuae