pibooth/pibooth #174: Aucune prévisualisation

babou7635, 2020-08-29:
Hello les amis,

J'ai installer la toute dernière version du pibooth. Je le run sans soucis, il fonctionne correctement mais hélas je n'ai aucun écran de prévisualisation. L'application va jusqu'au bout. J’obtiens la ou les photos demandées. Mon soucis est qu'il n'y a aucune prévisualisation alors que le timer est bien réglé sur 5 sec dans le menu de configuration. J'ai donc un écran noir pendant 5 sec et après il m'affiche ma photo sans problème.
J'ai aussi mis la fonction "Preview Countdown" sur True mais malheureusement je n'ai aucun affichage d'un décompte.

Est ce normal ? 

Merci de votre aide par avance car là je sèche ?

Merci les amis

**Environment**
 - OS [Raspbian Buster]
 - Camera [Rpi camera]
 - Pibooth version [2.0.0]
 - Pigame version [1.9.6]

anxuae, 2020-09-07:
Hi @babou7635 ,

No, this behavior is not normal.

Can you tell me if you are using a DSLR camera also (hybrid mode) ?
What kind of screen are you using?

babou7635, 2020-09-09:
Hello,
Thanks for your feedback.
I use the camera for the raspberry (Pi Noir Camera V2) only this one.
For the moment I use my raspberry connected in VNC. I don't have connected a screen directely to the raspberry.
I test with a screen to HDMI tomorow.
Tell me if i must to change my camera V2.

anxuae, 2020-09-09:
Hello @babou7635 

VNCViewer is the cause. The preview is an overlay managed at GPU low level, thus it is not accessible using the VNC protocol.

See the post: https://raspberrypi.stackexchange.com/questions/29537/sending-raspberry-pi-camera-preview-to-a-laptop-running-vnc-viewer .

babou7635, 2020-09-10:
Hello,

I use to HDMI via a screen and it's ok for me. All works !! It's very good.

Thanks ;)

closed 2020-09-10 by babou7635