pibooth/pibooth #176: NO liveview using (mirrorless) DSLR camera

Eddy-1980, 2020-09-08:
When using Pibooth in combination with mirrorless DSLR Sony A5100 there is NO liveview.
When using Pibooth in combination with mirrorless DSLR Canon M50 there IS liveview.

I add 2 logfiles of both camera's create with pibooth-diag command.

The Sony A5100 is missing the "viewfinder" parameter.

link to files [https://1drv.ms/u/s!AnOp5nAVFWjRiX_VxbB5aCoZU4gK?e=21FuOo](url)

anxuae, 2020-09-08:
Hi,

I found this issue in libgphoto2 : https://github.com/gphoto/libgphoto2/issues/441 which is related to this topic.

I will have a look how to use the `GP_OPERATION_CAPTURE_PREVIEW` ability instead of `viewfinder` parameter to check if a camera is compatible for preview.

werdeil, 2020-09-11:
@Eddy-1980 can you try to install this dev version?
- First remove the version of pibooth you have:
>sudo pip3 uninstall pibooth
- Then install a dev version following [these instructions](https://github.com/pibooth/pibooth/blob/master/docs/dev.rst)
- Finally checkout the MR version and test:
>git checkout gphoro2-preview
>pibooth

commit fa7212fd (2020-09-23): Fix #176: manage GP preview detection with GP_OPERATION_CAPTURE_PREVIEW (#178)

* Manage GP preview detection with GP_OPERATION_CAPTURE_PREVIEW

* Update diagnostic script
closed 2020-09-23 by anxuae