pibooth/pibooth #180: Photo Galery

TheSuperpapi, 2020-09-17:
Hello to all of you,

After spending time to read the requests hystories, I saw that you are a lot to be interesting for photo gallery.

I spent a long time on my own side to define what could be the best solution.
- By implementing  it, directly into the pibooth software
- By having an external solution (cloud upload)

Finally, I decided  to do it externally and for many reasons:

- I don't have no coding knowledges. 
  I only have the capacity to read, understand and personalize something already done 

- I wanted to have this galery accessible fro internet

- I wanted to have the choice to use my printer directly with the RPI or depending of the circumstance, to use it on a laptop connected to internet (allow people to print later)

--> Today I'll present you one of the existing solution (because I suppose that a lot are existing) by using rclone (https://rclone.org/)

To resume, rclone is able to synchronise folders between the rpi to an external source like GoogleDrive / OneDrive ... (check the list here: https://rclone.org/overview/ ** take care, amazon is no longer supported** )

I have spent my day to implement it on my RPI and here is the result on a smartphone view:

[image]

For me it's ok in term of gallery vs my need.
- We can see pictures
- We can open it in full screen
- We can download it
- We can print it with a laptop connected to the account (have to check if I can do that remotly with every device.... but one thing at one time)

So today, test was done on my side, now several issue have to solved to be effycient
1- Define the better way to automatize this synch in background task.

2- The "onedrive" folder share link is horrible... like this for exemple: https://1drv.ms/u/s!Al4kF940zW7mF90cLQS_aG8hS3M?e=aSEto

3- Be sure I have not wrong impact on Pibooth by doing that in background task

For sure that suppose your raspberry to be connected to internet, on standard connection or via smartphone connection sharing.

If some of us are interesting to push a bit more this way, I can support you for the first part (setting up rclone). 

Have a nice day and see you soon.
Papi

ttette, 2020-09-17:
Hi @TheSuperpapi 

Thanks for sharing your experience. 

In version 2.0, you can actually use directly a google photo plug-in that is integrated with the Pibooth 

https://pypi.org/project/pibooth-google-photo/

Setting it up is pretty easy - there are a bunch of tutorials how you get the necessary file from google photos.

Google photos worked great for me - I had a special Album created for the event and a QR code with the link next to the PhotoBooth, so all the guests could just scan it and access everything. I had the printer linked to the RPI, because I didn't want the guests to go crazy printing whatever they liked. Uploading was fast and I didn't have any issues with the connection using the plug-in.

TheSuperpapi, 2020-09-17:
Hello ttette,

Thanks for sharing your solution it'll give more choices to all.

If you like the Google photo interface, probably, this is à Good choice.

Does this plug in make à constant synchronisation to Google photo ?

anxuae, 2020-09-18:
Hello,

This plugin send the final picture to a Google Photo gallery, which is is done after each captures sequence.

Anxuae

TheSuperpapi, 2020-09-18:
Hello and thanks for the reply.

This is a really Good new to know that this is a picture to picture upload managment. 

This is interesting to know too, that it could be updated probably for onedrive for exemple.

For me google photo interface is not really friendly, but maybe it's just because I need to spend more time on. 

But this picture upload managment seems better compare to my idea. 
In my proposal i'm just able to ask the upload to be done in a certain periodicity [every 5 minutes  for exemple] 

I really need to spend more time on this plug in during next weeks after solving my crash issue. 

Thank again, very interesting to exchange with all of you.

TheSuperpapi, 2020-09-22:
Hello,

As promised I took a bit more time on the gallery topic.

It completely change my vision :).

In fact when creating a new account and sharing the link of the album, the picture view and interface is really better.

My first vision was wrong and base on my personal account merge with my phone data's, in that case it is not really convenient to find and see the right album.

But by creating a dedicated account without any synchronization and sharing the album link, it's completely different and just perfect.

So now, I think I'll stop my OneDrive account billing and maybe switch to google photo :)

Thanks,
Papi

closed 2020-09-22 by TheSuperpapi

TheSuperpapi, 2020-10-24:
Hello there,

Just a comment after some tests.
The google-photo plugin is great as I mention before but one think can switch your choice.

This pugin upload time to time each picture, that's songs good but take care.

If you have a low connection or the pi get discunnected, the pibooth softawere will stay frozen until the picture upload.

Depending of the use you decided to have with pibooth, I can recommand to use the first method I proposed (to insure that pibooth will always work fluently, and the picture will be or not uploaded externaly)

Maybe a delay of 15/20s is missing on thi google picture plugin (or maybe I don't see it).
Still need to spend time on.

Have nice day :-)