pibooth/pibooth #181: Small LCD display - Showing "Taken" Photos, from counters.

DJ-Dingo, 2020-09-18:
I am starting to build a new box for my PiBooth, and have a small LCD 16x2 showing Date-Time.
Would it be possible to have it show "Taken" Photos from Counters as well?
Or could it be a future plugin?

Example of what i am looking for to my 16x2 LCD (see picture):
Today Photos 197
19/09  -  01:20:53

Example on a 16x4 LCD could be:
Taken - 197
Printed - 00
Forgotten - 04
Remaining dub 03 

I am still waiting for and i2c for my LCD, to use less GPIO, But here is my code so far, for the 16x2 display

#!/usr/bin/env python3
import RPi.GPIO as GPIO
import time
from time import strftime
from RPLCD import CharLCD

lcd = CharLCD(cols=16, rows=2, pin_rs=36, pin_e=35, pins_data=[33, 18, 16, 23], numbering_mode=GPIO.BOARD)

lcd.clear()

while True:

    lcd.cursor_pos = (0, 0)
    lcd.write_string('Today Photos 197')
    lcd.cursor_pos = (1, 0)
    lcd.write_string('%s' %time.strftime('%d/%m - %H:%M:%S'))

GPIO.setwarnings(False)
GPIO.cleanup()

Love the PiBooth project, have had alot of fun with it. Keep up the good work.

[image]

anxuae, 2020-09-19:
Hello @DJ-Dingo ,

Don't this that it can be integrated in the pibooth code because it is generic enough.

However, you can write your own plugin to manage it, below an example that I have drafted (not tested):

```
"""Plugin to handle small LCD display."""

import time
import RPi.GPIO as GPIO
from RPLCD import CharLCD
import pibooth

__version__ = "1.0.0"

@pibooth.hookimpl
def pibooth_startup(app):
    app.lcd = CharLCD(cols=16, rows=2, pin_rs=36, pin_e=35,
                      pins_data=[33, 18, 16, 23], numbering_mode=GPIO.BOARD)
    app.lcd.clear()

    # Write the date at pibooth startup
    app.lcd.cursor_pos = (1, 0)
    app.lcd.write_string('%s' % time.strftime('%d/%m - %H:%M:%S'))

@pibooth.hookimpl
def state_wait_enter(app):
    # Re-write the number of taken pictures each time pibooth
    # enter in 'wait' state.
    app.lcd.cursor_pos = (0, 0)
    app.lcd.write_string('Today Photos %s' % app.count.taken)

@pibooth.hookimpl
def pibooth_cleanup(app):
    app.lcd.clear()
    GPIO.setwarnings(False)
    GPIO.cleanup()
```

Save it in a file, then put the path to this plugin in the `[GENERAL][plugin]` key of your configuration.

DJ-Dingo, 2020-09-20:
Thanks for the quick reply :-) And an example.

With a little change, I made it work. But the clock stands still at the same time as I started PiBooth, it does not run. But the Photo counter works fine.

I had to change my code a bit as it says "A different mode has already been set" and I can not use GPIO.BOARD since PiBooth uses gpiozero. So I changed the PIN number to use BCM instead.

But since I'll get a I2c tomorrow, I'll just test it out a little more, since I still have to change my code to run with the I2c.

But how do I get the clock to run, and not stand still.

werdeil, 2020-09-20:
Currently in @anxuae example the clock is only set at the startup (in `pibooth_startup` hook).
If you want to have the clock update in real time it will be a little bit difficult: you'll need to add the line for each state_<name>_do hooks (see explanation on hooks [here](https://github.com/pibooth/pibooth/blob/master/docs/plugin.rst))

If you manage to have a generic and working plugin we could integrate it in the pibooth project so everyone can use it.

DJ-Dingo, 2020-09-20:
Cool thanks.

Think I managed to make the clock work.
It jumps a bit in sec sometimes. But I can live with that :-)

I will post my code when I have finished testing it.
I also need to make a code that runs with I2c as well

DJ-Dingo, 2020-09-28:
[The plugin so far - Pibooth-LCD-with-I2c](https://github.com/DJ-Dingo/Pibooth-LCD-with-I2c)
Since I have already made it work (Thanks @werdeil and @anxuae ) and working on futures in another thread, I will close this one.

closed 2020-09-28 by DJ-Dingo